raspberrypi/documentation #1939: NVME guide assumes that the user is familiar with usbboot

pelwell, 2021-07-07:
@peterharperuk I've seen a number of forum posts where the user is running usbboot on a CM4 as a way of enabling NVME booting. The documentation here does not make it clear that usbboot/rpiboot must be run on a separate computer Pi/PC/MAC in order to update the CM4 boot EEPROM.

peterharperuk, 2021-07-07:
Thanks @pelwell. I haven't noticed these threads I'm afraid - which part of the forum should I be keeping an eye on? I've pushed a change that adds some clarification.

pelwell, 2021-07-07:
They end up in the [Compute Module forum](https://www.raspberrypi.org/forums/viewforum.php?f=98) most of the time, e.g. https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=315291, but there was one on GitHub: https://github.com/raspberrypi/usbboot/issues/82

lurch, 2021-07-07:
@peterharperuk Looks like you've pushed the change to your fork of the documentation - do you want to PR it here too? :wink:

aallan, 2021-07-07:
> …do you want to PR it here too? 

If you do @peterharperuk sooner is better, I'll probably be freezing the current Markdown-based documentation at the end of the week (see #1911 for details of what's going on there).

peterharperuk, 2021-07-07:
Sorry, PR is here https://github.com/raspberrypi/documentation/pull/1940

aallan, 2021-07-12:
PR is closed due to #1911.

closed 2021-07-12 by aallan