raspberrypi/documentation #1949: External storage configuration: automount

dkoehler69, 2021-07-19:
The page says:

By default, your Raspberry Pi automatically mounts some of the popular file systems such as FAT, NTFS, and HFS+ at the /media/pi/<HARD-DRIVE-LABEL> location.

This does not work for me.

OS: Raspbian GNU/Linux 10 (buster), 64bit kernel activated
External drive: Seagate Backup Plus, 2 TByte, NTFS

lurch, 2021-07-19:
ping @spl237  Does automount only work when you're running the graphical version of Raspberry Pi OS?

aallan, 2021-07-20:
We have fuse installed by default so yes it should work..?

spl237, 2021-07-20:
> ping @spl237 Does automount only work when you're running the graphical version of Raspberry Pi OS?

Check with @XECDesign - the lite version is his baby.

XECDesign, 2021-07-20:
No, lite does not have automount.

aallan, 2021-07-20:
> No, lite does not have automount.

Thanks. I'll add that as a note on the docs page.

@dkoehler69 looks like it's working as expected.

closed 2021-07-20 by aallan