raspberrypi/documentation #1951: Kernel building documentation cross compiling problem

sbela, 2021-07-21:
The kernel building documentation says that for 32 bit cross compiling use the "**crossbuild-essential-armhf**" compiler, but this compiler was not built for armv6, therefore **illegal instruction** exception occurs with a simple hello world test on my Raspberry Pi Model B Plus Rev 1.2

JamesH65, 2021-07-21:
Can you point to the documentation you believe is incorrect?

sbela, 2021-07-21:
In the documentation/linux/kernel/building.md at section **Cross-compiling** says to use the _sudo apt install crossbuild-essential-armhf_ for **Install the 32-bit toolchain for a 32-bit kernel** and I have a Raspberry Pi Model B Plus Rev 1.2 with:
uname -a: Linux pi1-2021 5.10.17+ #1421 Thu May 27 13:58:02 BST 2021 armv6l GNU/Linux
and lsb_release -a:
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

so, when I compile with the suggested compiler the following C code:
`int main(void)
{
  int i = 1;
  int j = i + 1;
  return 0;
}
`
with command line: $ arm-linux-gnueabihf-gcc -o test test.c -Wall 
it compiles fine, but when I run on the rapsberry pi it fails with an illegal instruction message.
Do I miss something ? Or I have made a mistake ?

lurch, 2021-07-21:
Those instructions are (only) talking about cross-building the Linux kernel, not about cross-building generic C code. Does the cross-built kernel also fail on your Model B Plus?

sbela, 2021-07-21:
I will test it today! Am I correct that there is no documentation for cross compilation for raspberry pi which is officially supported ?

aallan, 2021-07-21:
I don't believe there is any documentation around cross-compiing generic C code for Raspberry Pi.

lurch, 2021-07-21:
There's some very outdated toolchains at https://github.com/raspberrypi/tools/tree/master/arm-bcm2708 which _might_ do what you're looking for, but I suspect that they're so old that they're now "unsupported, use at your own risk".

sbela, 2021-07-21:
I tried the compiler for the kernel and it works! But I still do not know how to cross compile qt5 for pi1 armv6 without illegal instruction and c++17 support...

closed 2021-07-21 by sbela

lurch, 2021-07-21:
> But I still do not know how to cross compile qt5 for pi1 armv6 without illegal instruction and c++17 support...

That sounds like a question for the https://www.raspberrypi.org/forums/ rather than the documentation repo :wink: