raspberrypi/documentation #1959: bluetooth stopped working after latest update

0d0a, 2021-08-08:
when setting up a new pi400 i noticed that bluetooth works and connects to my bluetooth device(s) before i do the standard update
after the update bluetooth does not connect anymore. 
bluetoothctl says no default controller available 
uname -a 
5.10.52-v7l+ #1441 SMP Tue Aug 3 18:11:56 BST 2021 armv7l GNU/Linux

aallan, 2021-08-08:
This repo is for the Raspberry Pi documentaiton. This is the sort of problem you should ask about on the [forums](https://www.raspberrypi.org/forums/), so I’m closing this issue. But please follow up with further comments if you have specific questions about the [documentation](https://www.raspberrypi.org/documentation/).

closed 2021-08-08 by aallan

0d0a, 2021-08-08:
Sorry. Posted in the forums.
Thx

the white rider
(Het leuke van ... is weten op reis te zijn