raspberrypi/documentation #1973: Page location change has broken external links

andrum99, 2021-08-09:
And internal links within the website from parts outwith the documentation repo. For example:

https://www.raspberrypi.org/software/

All links on that page which point into the repo return a 404 error.

aallan, 2021-08-09:
Thanks, we thought we'd caught them all! We're fixing the links on the software page,

[image]

Is there anywhere else you can point to that 404's rather than being redirected?

NOTE: Some deep links will *by design* 404, but there shouldn't be any directly off the Raspberry Pi homepage(s).

andrum99, 2021-08-09:
Is there a redirect in place? If not, you probably should have one, since all links on 3rd party sites into the documentation pages will now be broken without one. It should be relatively straightforward to come up with a rewrite rule to do the redirect.

aallan, 2021-08-09:
> Is there a redirect in place? If not, you probably should have one, since all links on 3rd party sites into the documentation pages will now be broken without one. It should be relatively straightforward to come up with a rewrite rule to do the redirect.

There are redirects in place for all (hopefully) of the PDFs and other not-markdown files that were hosted as part of the documentation repo. However, we are not going to redirect all the deep links into the documentation to the new locations. 

That said, I totally agree the generic 404 page isn't helpful. We should generate a documentation-specific 404 page that is more helpful in directing users into the new documentation stack.

andrum99, 2021-08-09:
> Is there anywhere else you can point to that 404's rather than being redirected?

I've not checked - the 'software' page was the first one I hit when I was looking for something else. The web team may wish to avail themselves of an automated dead link checker, since upon running a free one online there seems to be quite a few internal links which have gone walkabout.

aallan, 2021-08-09:
> The web team may wish to avail themselves of an automated dead link checker, since upon running a free one online there seems to be quite a few internal links which have gone walkabout.

That's with the Foundation team. Not Trading.

andrum99, 2021-08-09:
> > The web team may wish to avail themselves of an automated dead link checker, since upon running a free one online there seems to be quite a few internal links which have gone walkabout.
> 
> That's with the Foundation team. Not Trading.

Indeed. I was kind of assuming you would have some way of communicating with the Foundation though.

andrum99, 2021-08-09:
What I meant was you could just use a single rewrite rule to point `raspberrypi.org/documentation/something` to `raspberrypi.org/documentation/computers/something, i.e. exclude raspberrypi.org/documentation/ itself from the redirect.

aallan, 2021-08-09:
We'll fix any internally hosted links we find — and pass any Foundation hosted links to our counterparts there — but for external links I think #1976 will be the solution.

closed 2021-08-09 by aallan

wolfd, 2021-08-10:
Is there a good place to report those dead links?

For now, here's two more:
- https://www.raspberrypi.org/blog/an-open-source-camera-stack-for-raspberry-pi-using-libcamera/
- https://libcamera.org/open-projects.html#learning-how-an-isp-is-used - not Raspberry Pi hosted, but a collaboration

aallan, 2021-08-10:
Links to individual `*.md` pages will not be redirected. We can't be expected to maintain, forever, a complex tree of links, and then redirect every page when the documentation is restructured, and then again when we restructure it the next time. Additionally, I'm about to embark on a large scale cull. My current estimate is about 1/3 of the existing documentation will probably be deleted. 

Link rot is a thing everybody has to deal with. 

However the link from the libcamera.org post you pointed at is to a PDF, those we are going to throw redirects in place for, in fact they should be there already — that one obviously got missed — we'll fix that one up.

aallan, 2021-08-10:
…we've also now added some category level redirects (see #2010) which should catch most 404'ed pages.