raspberrypi/documentation #1976: A documentation-specific 404 page is needed

aallan, 2021-08-09:
I think the best solution to #1973 is that the documentation site has its own 404 page which is more helpful; perhaps having the top-level categories? This will help redirect lost users back into the main documentation rather than just stranding them with a generic "404" message. 

[image]

Any ideas what should be on this page @jackbenwillis, @mudge ?

andrum99, 2021-08-09:
I wonder if it would be possible to have the documentation-specific 404 page just be the documentation index page itself? You could trigger some specific error message within the page by passing a parameter as part of the URL.

aallan, 2021-08-10:
We've now added some category level redirects (see #2010) which should catch most 404'ed pages. But a custom 404 page for the documentation is still needed I think.

aallan, 2021-08-10:
@jackbenwillis Can you have a think about what this should look like?

lurch, 2021-08-10:
https://web.archive.org/web/20140714193525/www.raspberrypi.org/404

aallan, 2021-08-10:
> https://web.archive.org/web/20140714193525/www.raspberrypi.org/404

The new 404 page should presumably direct people to the new documentation site, maybe let people search for things, and encourage them to update their dead links. Not entirely sure that helps! 😆

jackbenwillis, 2021-08-10:
[image]
What about this? Replacing the header with a 404 message and asking people to browse the updated documentation. This way you'll already be on the documentation landing page and be able to get going right away

aallan, 2021-08-10:
That'd work for me. Hopefully it'd be possible to configure it so that the landing page in the 404 isn't static or separately generated (and have to be updated independently from the main top page)...? @nelliemckesson @mudge ?

mudge, 2021-08-10:
> That'd work for me. Hopefully it'd be possible to configure it so that the landing page in the 404 isn't static or separately generated (and have to be updated independently from the main top page)...? @nelliemckesson @mudge ?

@nelliemckesson can confirm but I expect we can generate a separate 404 page much like we do the current index page, dynamically populating it with the same categories and even including the search functionality. I can then configure our web server to use that page whenever it encounters a 404 error rather than taking users to https://www.raspberrypi.org/404.html

aallan, 2021-08-10:
Let's do it this way then.

lurch, 2021-08-10:
@mudge I think we'd want to keep the existing generic https://www.raspberrypi.org/404.html page, and only use this new design fro 404s coming from the https://www.raspberrypi.org/documentation/ "subdirectory"? Is it possible to do that?

aallan, 2021-08-10:
>  Is it possible to do that?

I was presuming that's what he meant?

mudge, 2021-08-10:
Yes, that's what I meant. We won't touch https://www.raspberrypi.org/404.html as that is used by other parts of the site but instead configure the documentation server specifically to use our new documentation-only 404 page.

lurch, 2021-08-10:
Ah, sorry for getting my wires crossed :laughing:

lurch, 2021-08-10:
I just knocked up a quick prototype and came up with this:
[image]

Is everyone happy enough with that for me to create a PR?

jackbenwillis, 2021-08-10:
Could we increase the size of the '404'?

Otherwise looks good!

lurch, 2021-08-10:
I dunno how to change font-sizes, that's a CSS thing :laughing:  So I'll put together a PR with what I've got and @nelliemckesson can bump the font-sizes later :+1:

lurch, 2021-08-10:
@nelliemckesson I tried making "Raspberry Pi Documentation" into a link, but it didn't look right, so I won't do that for now :wink: 
[image]

aallan, 2021-08-10:
>  I tried making "Raspberry Pi Documentation" into a link, but it didn't look right, so I won't do that for now 

Eek. No, that's not right. 😬

lurch, 2021-08-10:
See #2026

jackbenwillis, 2021-08-10:
Yup! leave it out for now ;)

small thing, full stop after for and uppercase P in please (all my fault)!

"We couldn't find what you were looking for. Please browse our updated **Raspberry Pi** Documentation"

jackbenwillis, 2021-08-10:
and a full stop!

aallan, 2021-08-12:
@nelliemckesson If you could look at #2026 and sprinkle some CSS magic that'd be great! Thanks!

commit a8afd783 (2021-08-13): Merge pull request #2026 from raspberrypi/404_page

Add tooling to create a 404 page. Partially fixes #1976
commit cc8d8ea1 (2021-08-13): Add tooling to create a 404 page. Partially fixes #1976
closed 2021-08-13 by aallan