raspberrypi/documentation #1979: Left-hand navigation element width changed on hover (Firefox nightly)

Gadgetoid, 2021-08-09:
Perhaps best illustrated by a video: https://twitter.com/Gadgetoid/status/1424701341520310274

The left-hand navigation has `overflow-y:` changed from "hidden" to "auto" on mouseover with the intent of invoking the built-in scrollbar for overflowed elements.

Unfortunately the scrollbar appearing/disappearing changes the width of the element (using FF nightly 92.0a1) and can cause links to jump out from under the cursor when the user tries to mouse over them.

In Firefox `scrollbar-width: none` allows the element to be scrolled but removes the visible scrollbar, fixing this issue.

aallan, 2021-08-09:
Thanks, this isn't something that happens in Chrome or Safari. I guess none of us run Firefox. Will get this fixed.

lurch, 2021-08-09:
I _have_ seen this in Chrome - I think whether you see it or not just depends on whether your font-sizes and window-widths line up exactly enough to cause the scroll-bar-appearing to cause one of the titles to then word-wrap.

> but removes the visible scrollbar

IMHO we don't want to do that, as the user then has no hint that the left-hand navigation contains additional content that's off the bottom of the screen? Also, perhaps there are some people who have a mouse without a scroll-wheel, or a touchpad without a scroll-function? :shrug:

Gadgetoid, 2021-08-09:
> the user then has no hint that the left-hand navigation contains additional content

They already have no hint unless they mouse-over, so always-visible might be the way to go for clarity.

I feel for anyone without a scroll-wheel since these modern scroll-bars don't give you much girth to grab.

ReadTheDocs has no scrollbar and no hints there might be more elements :grimacing: : https://pillow.readthedocs.io/en/stable/reference/index.html

lurch, 2021-08-09:
I should probably add the caveat that these are only my personal opinions, and I'm not a professional UI or UX designer :laughing:

septatrix, 2021-09-01:
> Thanks, this isn't something that happens in Chrome or Safari. I guess none of us run Firefox. Will get this fixed.

I think this depends on the OS.
IIRC macOS uses overlay scrollbars (like on mobile phones) which do not occupy any space and therefore do not result in any layout shifts.

https://user-images.githubusercontent.com/24257556/131743901-4abda78b-a6ba-480e-a439-eb1ca3dffb89.mp4

septatrix, 2021-11-07:
This is still relevant and a nuisance. Interestingly this seems to be done on purpose(?!) as the style is set on the element dynamically using JS (not even in CSS which would also be possible) whenever one hovers over it.

lurch, 2022-01-12:
This is still an issue, so keeping this open.

JamesH65, 2022-03-21:
Still an issue

aallan, 2022-04-24:
I don't see this getting fixed.

closed 2022-04-24 by aallan