raspberrypi/documentation #1982: Lack of clear delineation between sections

Gadgetoid, 2021-08-09:
The font-size, style and spacing of headings and subheadings is close enough that it's very difficult to visually parse when a given section has come to an end and the next one begins:

[image]

While I'd normally very strongly favor individual pages per topic ( the docs in their current state *require* Javascript to be usable and... y'know what us nerds are like :grimacing: ) I think the current situation can be improved:

[image]

As an aside- is the current format of docs readily translatable into PDF format or already available as a PDF?

aallan, 2021-08-09:
> As an aside- is the current format of docs readily translatable into PDF format or already available as a PDF?

Potentially, yes. Although not immediately. Now the documentation is in Asciidoc, we can now potentially run it through the same toolchain as the RP2040 documentation stack and generate both PDFs and ePub versions. This was one of the goals of the [docs-ng project](https://www.raspberrypi.org/blog/bring-on-the-documentation/) and is one of the reasons for the structural changes — although there are others — hopefully this should happen (i.e. I get time free) to do this very soon.

lurch, 2021-08-09:
> when a given section has come to an end and the next one begins

I think that's just an artefact of the way that we've built these initial pages by ramming-together content from the old site, and I assume that things will flow more smoothly (so the start / end of sections will be less jarring) once @aallan has sprinkled his editing magic?

aallan, 2021-08-09:
> ...once @aallan has sprinkled his editing magic?

You have no idea how much editing magic has already gone on. 😄 

But yes, from here (see the [blog post](https://www.raspberrypi.org/blog/bring-on-the-documentation/)) we're embarking on what is essentially a total rewrite of the documentation. But I figured folks would want me to do that in public, not in private.

aallan, 2021-08-09:
I'm unconvinced by the big blank gap approach. But agree that the divisions between *major* section headings might be more clear. Let me think about it for a bit.

aallan, 2021-08-13:
I think a combination of #1991 and #2029 will probably fix most of this as its essentially a navigation issie.

closed 2021-08-13 by aallan