raspberrypi/documentation #1983: 'Using Linux' section should be removed

andrum99, 2021-08-09:
See #1972. I was assuming your pruning and re-writing was going to remove this section, so when I saw this process had not yet begun, I thought this would be an easy one to start us off with 😁

aallan, 2021-08-09:
> See #1972. I was assuming your pruning and re-writing was going to remove this section, so when I saw this process had not yet begun, I thought this would be an easy one to start us off with 😁

See the [blog post](https://www.raspberrypi.org/blog/bring-on-the-documentation/), I did what I considered the bare minimum. There is *a lot* of documentation that has already gone away. You just haven't noticed it yet (queue the screaming from some quarters).

I did think about dropping this entire section pre-launch. But figured I'd like to do it in public if we're doing it.

lurch, 2021-08-09:
We've opened the curtains :wink: 

[image]

andrum99, 2021-08-09:
> **Update:** The really interesting bit is where I'm going to be a lot more picky going forward about what we include. Picky means that things I don't want to add get a "no" and a closed PR.

I definitely approve of this approach 👍 Having much more focussed documentation is certainly a Good Thing &reg;

aallan, 2021-08-09:
> I definitely approve of this approach 👍 Having much more focussed documentation is certainly a Good Thing ®.

Right this. Hopefully, things will grow a bit less organically now I'm around and officially there in charge of it all. Today's launch should absolutely be seen as a "first step" not a "we're done now".

mi-hol, 2021-08-12:
Sorry I fail to understand the benefit of removing 'Using Linux' section for the RPI user community.
Many (if not most) new users of a RPI have never used a Linux OS before.
I personally found the RPI documentation the best and consistent documentation for learners.
Please elaborate on the planned approach before "killing" great content

andrum99, 2021-08-12:
There's plenty of better content out there, including on the MagPi website and in their PDFs. The documentation repo is being refocussed on Raspberry Pi-specific technical documentation and is no longer intended as a source of learning materials. Much of the content in this repo originated before the Raspberry Pi Foundation created the large range of learning content they are responsible for. The foundation now has a trading subsiduary which is responsible for the technical documentation. The trading subsiduary also produces the magazines and books.

Having said that, I feel it would be of benefit to users to have clear links to the foundation's c

closed 2021-08-30 by andrum99