raspberrypi/documentation #1984: Rendering: code sections always have a scroll bar at the bottom

andrum99, 2021-08-09:
Best illustrated by an example:

[image]

While it is certainly useful to have a scroll bar to cope with long lines, particularly on narrow displays, e.g. on mobile phones, I would personally prefer it if they did not appear all of the time.

aallan, 2021-08-09:
That's interesting. What Browser+OS are you seeing that in?

andrum99, 2021-08-09:
> That's interesting. What Browser+OS are you seeing that in?

Chrome version 92.0.4515.131 (Official Build) (64-bit) on Windows 10. I've got scaling set to 125% as it's a 15" laptop screen that's 1080p, which is the default for that device. I've not tried with scaling set to 100% but I would not expect that to make any difference.

aallan, 2021-08-09:
Thanks. We'll take a look. Not seeing that on either Chrome or Safari on Mac. 

[image]

lurch, 2021-08-09:
I get the scrollbar in Chrome 92.0.4515.107 on Ubuntu Linux
[image]

aallan, 2021-08-09:
This is where @nelliemckesson and I both being on Mac becomes a problem. 😆

andrum99, 2021-08-09:
> This is where @nelliemckesson and I both being on Mac becomes a problem. 😆

[image]

I used browsershots.org back in the day, but unfortunately this seems to be no more 😢

DeeJay, 2021-08-09:
Just as another data point, I also see the redundant horizontal scroll with MS Edge Version 92.0.902.67 (Official build) (64-bit) on MS Windows 10 Home - 
Version	20H2
Installed on	‎05/‎11/‎2020
OS build	19042.1110
Experience	Windows Feature Experience Pack 120.2212.3530.0

nelliemckesson, 2021-08-18:
@aallan what's the preferred behavior here? Scrollbar only visible for elements that actually overflow? Scrollbar only visible when hovering over an element that overflows? Scrollbar never visible?

aallan, 2021-08-18:
> @aallan what's the preferred behavior here? Scrollbar only visible for elements that actually overflow? Scrollbar only visible when hovering over an element that overflows? Scrollbar never visible?

Too many choices 🧠 

Umm. I **think** I'm going to opt for scrollbar only visible for elements that actually overflow, to give the visual cue that you can scroll, although personally I'd lean towards scrollbar only visible when hovering over an element that overflows, I think that might not be obvious to some users.

lurch, 2021-08-18:
"scrollbar only visible for elements that actually overflow, to give the visual cue that you can scroll"

If we could do the same for the ToC scrollbar, that'd fix #1979  :wink:

commit 61ed09b1 (2021-09-06): attempting to fix #1984
closed 2021-09-06 by aallan