raspberrypi/documentation #1987: Search bar only returns the first 10 results

aallan, 2021-08-09:
The built-in search bar will return only the first 10 results and there is no visual indicator that only the first 10 results are being shown…

[image]

…need to think about how to work around this UI/UX problem.

andrum99, 2021-08-09:
I hadn't clocked that the drop-down list of pages wasn't just a preview. Based on how Chrome works, I was expecting the list that appears as you type to just be a preview, then the actual proper search would then take place once you pressed enter or clicked on the search button.

aallan, 2021-08-13:
> I hadn't clocked that the drop-down list of pages wasn't just a preview. Based on how Chrome works, I was expecting the list that appears as you type to just be a preview, then the actual proper search would then take place once you pressed enter or clicked on the search button.

This might be the solution to carrying a full index of the site around on all of the pages. We can carry our current (fairly small) index around, and then if the user pushes "Enter" then we do a full text search (rather than a section heading search)? Maybe? Thoughts from @lurch, @mudge, and @nelliemckesson welcome here about how practical that is?

lurch, 2021-08-13:
I'm afraid "adding a search function to a site" is something I have no familiarity/experience with! But I'm happy to add any toolchain / index-generation support code that might be needed for that.

I guess if we're adding a "press `Enter` to do a full-search" feature, then we ought to make that more explicit / obvious by adding a `Search` button to the RHS of the search-box? (which would do the same thing as pressing `Enter`)

aallan, 2021-08-13:
> I'm afraid "adding a search function to a site" is something I have no familiarity/experience with! 

Yeah, I think that bit is more of a @nelliemckesson or perhaps @mudge thing. It's possible we could third-party it and use a Google Custom Search Engine — the look&feel of the return results can be customised a fair bit by CSS.

> But I'm happy to add any toolchain / index-generation support code that might be needed for that.

Thanks!

> I guess if we're adding a "press `Enter` to do a full-search" feature, then we ought to make that more explicit / obvious by adding a `Search` button to the RHS of the search-box? (which would do the same thing as pressing `Enter`)

Yup.

andrum99, 2021-08-13:
> It's possible we could third-party it and use a Google Custom Search Engine — the look&feel of the return results can be customised a fair bit by CSS.

That would be my preference. It means you're not adding extra load to the webserver by running the search using dynamic content on your own server. Most websites with their own search engine seem to do a very bad job of returning the most relevant results, and the forums also had trouble with their search: the recommended workaround was just to use Google with `site:forums.raspberrypi.org`.

aallan, 2021-08-13:
The downside of using Google would of course be that, in the short term, the index of `raspberrypi.org/documentation` would be pretty poor, although I think you can force a reindex if you use Google's Custom Search.

lurch, 2021-08-13:
I've suddenly remembered that the last time I did anything web-search related was creating an XML sitemap - in Perl :rofl:

aallan, 2021-08-13:
> I've suddenly remembered that the last time I did anything web-search related was creating an XML sitemap - in Perl 🤣

Yup. The last time I did search engine stuff I was writing in Perl. 😆

nelliemckesson, 2021-08-18:
I've never setup a Google Custom Search Engine, but happy to take a stab at it! Doesn't look to be terribly difficult, based on what I'm seeing with some initial googling. What's the timeframe/priority on this? Also, just to get the expected results written out on paper:

- User should be able to type in the search box and see an initial list of 10 results
- User should be able to press Enter/Return on their keyboard and be taken to a separate page of complete, full-text results
- User should be able to push a button in the UI and be taken to a separate page of complete, full-text results
- The page of complete results should ideally include the following elements/design: (Things like a link back to the main index page, etc... No idea right now what's actually feasible with this, but @jackbenwillis @aallan go ahead and insert a wishlist of stuff here and I'll do some more research/testing let you know if it's possible)

aallan, 2021-08-18:
> I've never setup a Google Custom Search Engine, 

I've done it a couple of times, but never customised the CSS that much.

> but happy to take a stab at it! Doesn't look to be terribly difficult, based on what I'm seeing with some initial googling. 

My guess is it'll be easy until it's not, there will be something that we want in CSS/stylistic terms that should be simple but will be increadibly difficult, because Google.

> What's the timeframe/priority on this? Also, just to get the expected results written out on paper:

@jackbenwillis is on holiday right now, and I'd like for him to mock up a search results page before you do too much heavy lifting.

> * User should be able to type in the search box and see an initial list of 10 results

Yes.

> * User should be able to press Enter/Return on their keyboard and be taken to a separate page of complete, full-text results

Yes. Needs to be some sort of visual indicator that this is a thing

> * User should be able to push a button in the UI and be taken to a separate page of complete, full-text results

Maybe? Google main page doesn't have a button to search any more (although it did originally, back in the day)

> * The page of complete results should ideally include the following elements/design: (Things like a link back to the main index page, etc... No idea right now what's actually feasible with this, but [...]

Let me think about this for a bit.

andrum99, 2021-08-18:
> Yes. Needs to be some sort of visual indicator that this is a thing
>
>
> Maybe? Google main page doesn't have a button to search any more (although it did originally, back in the day)

I'm not sure that you need a visual indicator that you can press enter if there is also a search button to press, since that tells the user that there is a further action which can be taken beyond just typing in the box. Perhaps a magnifying glass icon on the righthand end of the text input box?

lurch, 2021-08-18:
> Google main page doesn't have a button to search any more

It does?
[image]

aallan, 2021-08-18:
> > Google main page doesn't have a button to search any more
> 
> It does?

Interesting, not for me in Chrome. Although it does in Safari.

[image]

lurch, 2021-08-18:
I think you're confusing your browser's "new tab" page, with the Google home page?

aallan, 2021-08-18:
I actually think I've got my browser set up so I literally can't get to the Google home page any more, it redirects things to a custom skinned new page. Interesting. I, err, hadn't realised I'd done that!

nelliemckesson, 2021-09-14:
Ping @jackbenwillis re: UI design for this. I'm in no rush, but figured I'd poke people a bit as long as I'm doing raspberry pi things!

jackbenwillis, 2021-09-14:
Thanks @nelliemckesson ! I'll have a look soon

aallan, 2021-11-14:
This still needs fixing at some point.

lurch, 2021-12-06:
@timg236 asked if the documentation search function can search for keywords instead of just searching the titles / headers; I pointed him at this issue :wink:

aallan, 2021-12-06:
> @timg236 asked if the documentation search function can search for keywords instead of just searching the titles / headers

So it can, and the easiest way is to integrate and skin, a custom Google search. It's on the list.

closed 2022-02-01 by aallan

timg236, 2022-02-01:
Looks great and is very useful for searching for settings. When I tried searching for 'BOOT_UART' it never returned the EEPROM config page, but every other keyword there seemed to match!