raspberrypi/documentation #1990: Compute Module pages all giving 404s today

geerlingguy, 2021-08-09:
I've been trying to find Compute Module documentation pages today but it seems they're all 404'ing (maybe due to some reworking of the URL structure?).

For example:

  - https://www.raspberrypi.org/documentation/hardware/computemodule/cmio-camera.md
  - https://www.raspberrypi.org/documentation/hardware/computemodule

For now I've had to resort to Google's cached versions.

If there is a new URL structure, is it possible to add redirects? I was clicking through some 3rd party guides and they were linking to these pages where I kept getting 404s.

aallan, 2021-08-09:
See the [blog post](https://www.raspberrypi.org/blog/bring-on-the-documentation/) for details.

closed 2021-08-09 by aallan

aallan, 2021-08-09:
For external links I think #1976 is the solution we're going for.

aallan, 2021-08-10:
…and we've also now added some category level redirects (see #2010) which should catch most 404'ed pages.