raspberrypi/documentation #1991: Add anchor links for sections in the documentation

geerlingguy, 2021-08-09:
If possible, it would be a huge improvement to have permalink icons appear on hover over any section heading (since they're already set up as a named anchor in code), and when the user clicks on it they can get the permalink to that section of the page:

[image]

It doesn't matter to me whether on the right or left, but here's what I'm talking about (example from GitHub's default README layout, in this case on this repo's readme (https://github.com/raspberrypi/documentation):

[image]

andrum99, 2021-08-09:
On a related note, with such long pages it would be useful to have a "go to top of page" button somewhere.

lurch, 2021-08-09:
a) There's not (yet) a way to get the anchor from the titles themselves, but you _can_ get the anchor links from the Table-of-Contents on the left.

b) There's not (yet) a "go to top", but if you're looking at e.g. https://www.raspberrypi.org/documentation/computers/raspberry-pi.html then you can click on "Computers" at the top of the Table-of-Contents to go directly to https://www.raspberrypi.org/documentation/computers/

DeeJay, 2021-08-09:
I came to make the request already raised by @geerlingguy .

So a 'me too.'

But referring to Andrew @lurch - "you can get the anchor links from the Table-of-Contents on the left."  Under what circumstances? With Edge under Win10 a standard RightClick doesn't work. [Workaround is to 'screenscrape' from the address bar at the head of the browser window: that does contain the 'anchor-fied' url.]

lurch, 2021-08-09:
@DeeJay Oh yeah, I never noticed that :confused:   It must be a side-effect of http://gregfranko.com/jquery.tocify.js/ (which is what we use to underline the section-titles in the ToC as you scroll through the main body-contents).

aallan, 2021-08-09:
We like this suggestion to the extent that we're going to implment it. It's in the queue.

closed 2021-08-18 by aallan