raspberrypi/documentation #1998: Linking, Terms and Conditions

davidcoton, 2021-08-09:
From the Ts&Cs (https://my.raspberrypi.org/terms-conditions?approval):

    12.1 You may link to our home page or pages within the site,** provided you do so in a way that is fair and legal and does not damage our reputation or take advantage of it.
    ...
    12.3 Our site must not be framed on any other site, **nor may you create a link to any part of our site other than the home page.
This is contradictory.

aallan, 2021-08-09:
Ping @Roger-Thornton.

andrum99, 2021-08-09:
Reproducing my comment from the forum (at https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=317408&p=1899873#p1899764):

> 12.3 seems to be just very badly worded. I read it as attempting to say you must not use an IFRAME to embed Raspberry Pi official content within a 3rd-party website. The idea being this would make it look as though Raspberry Pi was associated with the 3rd-party website. 12.1 explicitly says that "you may link to our home page or pages within the site", which contradicts the bit that says "don't link to anything other than our home page" wording on 12.3.