raspberrypi/documentation #2000: Power Requirements documentation

davidcoton, 2021-08-09:
Power supply info https://www.raspberrypi.org/documentation/computers/raspberry-pi.html#power-supply

We have the typical consumptions, but we've lost the second table with the maximum (stress) ratings -- which is (was) far more useful when answering questions about suitable power supply arrangements.

andrum99, 2021-08-09:
I've looked at the history for this page (at https://github.com/raspberrypi/documentation/commits/ee338cb743f53a08df53d64f4072a889ec0c6617/hardware/raspberrypi/power) and I don't see any second table. Can you provide a link to the page you are referring to?

davidcoton, 2021-08-09:
It is no longer accessible. I think the link was https://www.raspberrypi.org/documentation/faqs/#pi-power, but it belonged in the power section, and the FAQs seem to have vanished.

aallan, 2021-08-09:
I can't see any evidence of a second table on the corresponding Markdown page?

aallan, 2021-08-09:
> …and the FAQs seem to have vanished.

The FAQ was very out of date and has been deleted. That happened some time ago, well before today's switchover.

davidcoton, 2021-08-09:
This table in the EAQs was not out of date, and was useful. I had it open in a browser tab until today, when part of the path suddenly became 404.
Please find the information and put it in the new documentation. Help us to help you by answering questions on the forum with authoritative references.

aallan, 2021-08-09:
> Please find the information and put it in the new documentation. 

The table doesn't exist on the page you're suggesting it should exist at, so unless you can point me in another direction I'm at a bit of a loss. There isn't any evidence in the Git history that that page ever had 2 tables.

andrum99, 2021-08-09:
> > Please find the information and put it in the new documentation.
> 
> The table doesn't exist on the page you're suggesting it should exist at, so unless you can point me in another direction I'm at a bit of a loss. There isn't any evidence in the Git history that that page ever had 2 tables.

I've found it one an old copy of the FAQ page - see
  http://web.archive.org/web/20210126005745/https://www.raspberrypi.org/documentation/faqs/#pi-power-specs

aallan, 2021-08-09:
> I've found it on an old copy of the FAQ page - see
http://web.archive.org/web/20210126005745/https://www.raspberrypi.org/documentation/faqs/#pi-power-specs

Ah! Yup. That's on my list of things to do, go through the FAQ and pick out the information in it that wasn't out of date, and insert it into relevant places in the documentation. I'll take this as a vote for the max current draw table. 😄 

See this PR for a copy of the FAQ just before it was deleted, https://github.com/raspberrypi/documentation/pull/1913. Looking at the PR that table came up there as well. I even have a post it note about it stuck to my monitor!

aallan, 2021-08-09:
I've gone ahead and added that section of the FAQ back into the documentation, see commit 68a1866. It's sitting on the `develop` branch at this point, and will go live onto the main site next time I merge to `master` which will push to production.

closed 2021-08-09 by aallan