raspberrypi/documentation #2013: ToC non-functional with Javascript disabled

lurch, 2021-08-10:
In #1935 @Gadgetoid complains that the new site is "a little bit completely broken without Javascript" which I was kinda surprised by.
So I gave it a try myself with JS turned off, and it seems that https://www.raspberrypi.org/documentation/ works okay (with the exception of the search box, because that's entirely implemented by javascript), https://www.raspberrypi.org/documentation/accessories/ works okay, and the _contents_ of e.g. https://www.raspberrypi.org/documentation/accessories/display.html works okay.
AFAICT it seems that the only bit that doesn't work is the exapanding and collapsing of sections in the Table of Contents on the left? _Perhaps_ we should tweak things so that **if** JS is disabled, the ToC is fully-expanded rather than being fully-collapsed?

aallan, 2021-08-10:
Thoughts @nelliemckesson ?

Gadgetoid, 2021-08-10:
FWIW the page I landed on was this - https://www.raspberrypi.org/documentation/computers/configuration.html

And the left-hand navigation looked "complete" ish (I wasn't aware of the drop down menu at this point) but clicking the links did nothing so it felt like I'd been dropped on a looooonnng page with now way to navigate it. Yes... I'm the kind of difficult pedant who browses the web with javascript disabled by default.

As @lurch suggests, I'd recommend a "progressive enhancement" approach whereupon the submenus are always enabled - ie: not hidden by default with the CSS - and only hidden by javascript at runtime. This *usually* happens fast enough that you don't get a surprising flash of expanded menus.

style.css line ~207 shows these submenu items require a "hidden" class in order to be hidden, so just removing that from the HTML and adding it at runtime with JS would suffice:

```css
nav#contents .itemcontents.hidden,
nav#mobile-contents .contents-inner.hidden {
  height: 0px;
  overflow: hidden;
}
```

(Granted the main thrust of web development experience is at least 5 years out of date so I could be spouting nonsense.)

closed 2021-09-14 by aallan

lurch, 2021-09-15:
ping @Gadgetoid The fix for this is now live at https://www.raspberrypi.org/documentation/ :slightly_smiling_face:

PR #2154 (2021-09-14): Adding non-JS TOC
@aallan this ought to fix #2013 . I've used the proposed solution to apply the `hidden` class on load via JS, and also tweaked some other JS bits so that the TOC will be scrollable, etc. Note that this will probably nullify the weird-scroll-issue fix, but only if JS is turned off (seems like an acceptable sacrifice to me).
commit 2a58a7c5 (2021-09-14): fixes #2013