raspberrypi/documentation #2029: Links to "Edit this page on Github"

aallan, 2021-08-11:
We potentially need more links back from the documentation site to the Github repo. 

I was thinking we might put an "Edit this page on Github" link in a (smaller) font (and maybe in italics) just under every level 2 header "==". Most (all?) level 2 headers are used as top-of-adoc file headers so it'd be an easy link back from the section to the specific asciidoc file that's contains the text. 

Might (will!) needs some toolchain magic perhaps to insert links automatically (or alternatively I could just go through and add them all manually, but that sounds like a bit of a nightmare to maintain long term.

Thoughts @lurch @nelliemckesson ?

lurch, 2021-08-11:
Sounds reasonable, but I dunno if it'd be "too distracting" having an "Edit this page on Github" link under _every_ level 2 header? What does @jackbenwillis think?

Yeah, we'd definitely want to automate this via the toolchain, shouldn't be _too_ hard to add :slightly_smiling_face:

aallan, 2021-08-11:
Level 2 headers are normally the header used a the top of an Asciidoc file. They aren't used (much) further down where it switches to level 3 for the most part. If we make it smaller font (lighter colour) and italics, it might not be *too* intrusive.

aallan, 2021-08-11:
> Yeah, we'd definitely want to automate this via the toolchain, shouldn't be too hard to add 🙂

Anyway, if you could have a look at it and see how hard it'f be I'd appreciate it.

lurch, 2021-08-11:
I've just knocked up a quick hand-edited prototype, and it looks like this. The links work as you'd expect them to, i.e. the second link goes to https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/os/updating.adoc . It obviously needs a bit of Nellie-CSS-magic sprinkled over the top, but I'll go ahead and code this up into a working solution... :+1: 
@nelliemckesson as these links are all auto-generated, let me know if you want some extra attributes/metadata/etc. added to them so that they'll be easier for your CSS to differentiate from regular links.
@aallan Maybe it should say "edit this section" rather than "edit this page"? :man_shrugging: 
[image]

aallan, 2021-08-11:
Oh that's excellent! Thanks @lurch. 

Maybe just `Edit this <a href="foobar">on Github</a>` and only make the "on Github" the link?

Definitely needs some CSS magic — make it smaller font, maybe italic, and closing up the whitespace and getting it to hug the Level 2 header pretty closely underneath?

commit d2af7a67 (2021-08-13): Add "Edit on GitHub" links to each page / section. Fixes #2029
commit 70a1d1b6 (2021-08-18): Merge pull request #2072 from raspberrypi/edit_this_page

Add "Edit on GitHub" links to each page / section. Fixes #2029
closed 2021-08-18 by aallan