raspberrypi/documentation #2048: README.md gives build instructions, doesn't say what the repo contains

andrum99, 2021-08-11:
Post-changeover the README.md file for this repo gives instructions for how to build HTML pages from the files in the repo, but it does not say what the repo actually contains. The solution is to move the build instructions into BUILDING.md and restore the explanatory text from the old README.md, modifying it to match the new reality.

aallan, 2021-08-11:
> restore the explanatory text from the old README.m

The top-level README.md on the old repo was effectively the main index page of the documentation. It did a poor job of explaining what was here, because it was indexing it, not talking about it. While I agree we probably want a short paragraph explaining what the repo contains at the top of the current README.md, I think the main point of the README file should be to talk about building the "software" the repo contains. This repo is source code, the binary is the HTML site, and that (and the real documentation site) is elsewhere. This isn't the documentation, this is the source of the documentation.

lurch, 2021-08-11:
> you don't need to go to the effort of setting up a build and running it to contribute PRs to the documentation.

That'll hopefully be made more obvious by #2029 ?

But yeah, the build-instructions part of the README will be getting an overhaul anyway next week.

aallan, 2021-08-13:
I edited the top of the file to at least say this was the documentation repo, and to point to the live site. But README.md is scheduled for an overhaul next week in any case.

closed 2021-08-13 by aallan