raspberrypi/documentation #2057: Image loading Issue

aallan, 2021-08-12:
This is an issue that @davidplowman is also seeing. 

I'm loading in images as normal, e.g.

```
=== Boards from Raspberry Pi

image::images/pico.jpg[]

https://www.raspberrypi.org/products/raspberry-pi-pico/[Raspberry Pi Pico]:: Designed by Raspberry Pi, RP2040 features a dual-core Arm Cortex-M0+ processor with 264KB internal RAM and support for up to 16MB of off-chip Flash. A wide range of flexible I/O options includes I2C, SPI, and — uniquely — Programmable I/O (PIO). These support endless possible applications for this small and affordable package.
```

but the images aren't displaying. 

[image]

I'm getting the following error in console when looking at things locally using `make serve_html`,

```
[2021-08-12 16:00:19] ERROR `/documentation/microcontrollers/images/pico.jpg' not found.
```

and checking in `documentation/html` in the relevant place the images aren't being copied over. Yet other images, loaded from the same directory, are being copied over. So images on that path are being found. Tried with both JPG and PNG images in case there was something weird going on there, but there isn't.

aallan, 2021-08-12:
See the associated PR #2058, which illustrates this issue.

aallan, 2021-08-12:
I'm really unsure what's happening differently here than elsewhere in the documentation. This image,

```
documentation/asciidoc/microcontrollers/rp2040/rp2040_explanation.svg
```

is loading from the same directory from `about_rp2040.adoc` as the `.jpg` and `.png` images in the same directory that **aren't** correctly loading from `rp2040_based_boards.adoc`.

closed 2021-08-13 by aallan

lurch, 2021-08-13:
Haha, this was a super-sneaky bug to find! It turns out that a faulty-regex meant that if there were any **include::** lines that _didn't_ end with a newline (i.e. the very last line in a file), then that included-file wouldn't get scanned for **image::** lines (and thus the build-system wouldn't know that those images needed copying).
Fixed by #2070

JamesH65, 2021-08-13:
I fixed this with the changes in white papers and PCN. Months ago. Assuming
this is the same issue. Thought my changes had been backported.

aallan, 2021-08-13:
> I fixed this with the changes in white papers and PCN. Months ago. Assuming this is the same issue. Thought my changes had been backported.

No. None of your changes to the white papers and change notice toolchains have been backported yet. 

At some point in the future, we need to consolidate the PDF toolchain(s) into its own repo and then sub-module them to all the repos using the toolchain, backporting changes from the individual repos will be done then. More ambitiously we need to merge the toolchain in this repo (Asciidoc->HTML) with the Asciidoc->PDF toolchain allowing us to generate sites and PDF/ePub from the same Asciidoc source. More work needed.

lurch, 2021-08-13:
> Assuming this is the same issue.

Nope, this isn't the same issue :wink: 
(But since your report with the inline-images in your PCN repo @JamesH65 , whenever writing new code I now always regex-check for `'image::?(.+?)\[.*\]'` to match both `image::blah[]` and `image:blah[]` :+1: )

> At some point in the future, .... More work needed.

More time needed. :rofl:

commit 8ff32bd5 (2021-08-13): Fixes #2057
commit b4dc27eb (2021-08-13): Merge pull request #2070 from raspberrypi/fix_scan_adoc

Fixes #2057