raspberrypi/documentation #2075: Question re style guide(s)

andrum99, 2021-08-14:
Before the switch to the new toolchain, this repo was meant to follow Raspberry Pi's own style guide, at https://github.com/raspberrypilearning/style-guide/blob/master/style-guide.md. Concurrent with the switch to the new toolchain, the link to Raspberry Pi's style guide was removed: this left things ambiguous. Then today aallan said:

> We generally follow the AP Style Guide, so we use title case in section headings. Kernel should be capitalised in this instance.

at https://github.com/raspberrypi/documentation/pull/2073#discussion_r688904669. This contradicts the previous policy, since there are a number of differences between the two style guides.

Please can we have the official policy on which style guide to use for this repo clarified?

aallan, 2021-08-14:
We're talking about this internally.

closed 2021-08-14 by aallan

andrum99, 2021-08-14:
Thank you for explaining that this is WIP. Since the issue itself still exists, I would prefer it if this issue was left open.

andrum99, 2021-08-20:
I've just noticed that the AP style guide - full name Associated Press Stylebook - is only available if you pay AP some money - either $26.95 for a hardcopy, or $22 per year subscription. Mandating the use of a style guide which is not available for free seems like a backward step to me.

aallan, 2021-08-20:
> I've just noticed that the AP style guide - full name Associated Press Stylebook - is only available if you pay AP some money - either $26.95 for a hardcopy, or $22 per year subscription. Mandating the use of a style guide which is not available for free seems like a backward step to me.

The AP style guide is the generally accepted standard in professional journalism.

andrum99, 2021-08-20:
> > I've just noticed that the AP style guide - full name Associated Press Stylebook - is only available if you pay AP some money - either $26.95 for a hardcopy, or $22 per year subscription. Mandating the use of a style guide which is not available for free seems like a backward step to me.
> 
> The AP style guide is the generally accepted standard in professional journalism, and to be fair, the "simplified" guide only costs $6 in paper back.

Thanks - I hadn't spotted the simplified guide. I've had another think about this and I think my objection is unreasonable. It is entirely proper that Raspberry Pi should use the style guide that they feel best suits their needs. Apologies. 

Anyway, this is an area that I am quite interested in, having done a wee bit of work on parts of the documentation, so I might just buy myself a copy of the full guide. (I don't think the simplified version will be detailed enough for my interests).

Going forward, I would imagine that someone at Raspberry Pi would probably fix up each contribution to be compliant with the style guide, and they would have a copy of the style guide to use. That's the impression I get from what Liz said anyway. In which case people like me who randomly contribute bits here and there to the documentation would not need to have their own copy of the style guide.

aallan, 2021-08-21:
Yup, and bear in mind that we don't *strictly* adhere to the AP style guide. It's more of a… guide?

lurch, 2021-08-22:
Are you saying we need a guide for how and when to apply the guiding principles from the AP style guide? :wink: :upside_down_face: