raspberrypi/documentation #2076: README.md and CONTRIBUTING.md - duplicate content

andrum99, 2021-08-14:
Build instructions for this repo are included in both files, and each has different wording (and perhaps method). Only one file should contain the build instructions, otherwise it makes things confusing and creates an unnecessary maintenance overhead.

aallan, 2021-08-14:
I actually can't see any build instructions in the `CONTRIBUTING.md` file, but I've been meaning to rewrite it anyway, see d1150d2.

closed 2021-08-14 by aallan

andrum99, 2021-08-14:
> I actually can't see any build instructions in the `CONTRIBUTING.md` file, but I've been meaning to rewrite it anyway, see [d1150d2](https://github.com/raspberrypi/documentation/commit/d1150d25078625576483364756f0d2d53a18b07a).

Apologies - I got that slightly wrong. In actual fact, CONTRIBUTING.md contained instructions for how to contribute to the repo, which is related to the build instructions but is not the same as them. I think today's rewrite is a big improvement.