raspberrypi/documentation #2079: Questions re copyright of contributions

andrum99, 2021-08-14:
The documentation in this repo continues to be made available via a Creative Commons license, specifically CC BY-SA 4.0 - this is the same licence used before the switch to the new toolchain. However, after the documentation built with the new toolchain was released, the following additional text appeared on `raspberrypi.org/documentation`:

> The Raspberry Pi Documentation is Copyright (c) 2012-2021 Raspberry Pi Trading (Ltd) and is [...]

This is surprising to me, since there was previously no indication that a contribution which I made to the documentation repository would be treated as being copyright of anyone other than myself. In fact, unless I assign copyright to a third party, I would expect my contributions to remain copyright of me. There is also no indication in this repo, other than in the copyright message added as an include within each page, that contributions to this repo will be treated as being copyright of Raspberry Pi (Trading) Ltd. (As an aside, you have put the parentheses in the wrong place - see above quote).

Can you possibly clarify what the copyright policy is for contributions to this repository which come from persons and companies outside the Raspberry Pi group of companies?

andrum99, 2021-09-13:
In the absence of any statement to the contrary, I can only assume that Trading is claiming copyright on all external contributions. In which case: cheerio.

JamesH65, 2021-09-14:
@aallan We really need to clarify this, Andrum99 was a long term helper and we've lost his welcome input. If we continue with this copyright claim we may as well just close the repo to third party contributions, as its seems unlikely people would be willing to accept it.

andrum99, 2021-11-03:
>people making a contribution here still hold the copyright to their contribution

That's what I thought should be the case.

>but because they've actively published that contribution based on our repo, which has SA in it, GPL-like magic happens.

I don't know what this means. 

I was questioning the sudden appearance of specific wording which implied that all content in the documentation repo was copyright Raspberry Pi (Trading) Ltd. If I, as an external contributor, make a contribution to the repo, and I retain copyright of that contribution, then logically the whole repo is not copyright Trading. Therefore, the text:

```
The Raspberry Pi Documentation is Copyright (c) 2012-2021 Raspberry Pi (Trading) Ltd
```

is misleading, since external contributors retain the copyright of the parts they wrote, which this fails to mention. Or have I missed something?

(I will also note that Wikipedia uses a slightly different licence, version 3.0 of the CC BY-SA licence, while this repo uses version 4.0, and that they don't have a blanket copyright message claiming the whole of Wikipedia is copyright the Wikimedia Foundation).

closed 2021-12-01 by aallan