raspberrypi/documentation #2080: Headless setup: No mention of "ssh" file

todbot, 2021-08-14:
The "Setting up a Headless Raspberry Pi" section at:
https://www.raspberrypi.org/documentation/computers/configuration.html#setting-up-a-headless-raspberry-pi
makes no mention of how to enable the SSH server. You have to know to go digging around in the "boot" folder section.

lurch, 2021-08-15:
The bottom of that section says "With no keyboard or monitor, you will need some way of remotely accessing your headless Raspberry Pi." and links to https://www.raspberrypi.org/documentation/computers/remote-access.html#remote-access which _does_ mention "For headless setup, SSH can be enabled by placing a file named ssh, without any extension, onto the boot partition of the SD Card. When the Raspberry Pi boots, it looks for the ssh file. If it is found, SSH is enabled and the file is deleted. The content of the file does not matter; it could contain text, or nothing at all." in the section describing enabling the SSH server.

(IIRC the https://www.raspberrypi.org/documentation/computers/configuration.html#setting-up-a-headless-raspberry-pi page **did** used to mention the `ssh` file, but I think it got edited out as it was considered duplicate content? :man_shrugging: )

It's up to @aallan to decide if this needs any improvement...

aallan, 2021-08-15:
Added a bit more meat to that `[NOTE]` section, see 9de73e3. Will go live next time I push to prod.

closed 2021-08-15 by aallan