raspberrypi/documentation #2089: Add asset-fingerprinting for CSS files

lurch, 2021-08-17:
As discussed in #2026 we could add a content-based hash (see e.g. some of the suggestions in https://stackoverflow.com/questions/4567089/hash-function-that-produces-short-hashes ) to the CSS filenames in order to work around the overly-keen (?) caching done by Cloudflare.

This could be done by e.g. calculating the hash of (the contents of) `boxes.css` as `abdef79` and then when the build-system copies files from `jekyll-assets/` to `build/jekyll/` it would rename `css/boxes.css` to `css/boxes-abdef79.css` and also modify `_layouts/boxes.html` to reference `{{ site.baseurl }}/css/boxes-abdef79.css` instead of `{{ site.baseurl }}/css/boxes.css`.

@aallan How often are we likely to be changing the CSS files? (i.e. how necessary is this change?)

@mudge Is it only `*.css` files that we need to do this for, or also other filetypes?
```bash
$ find documentation/html/ -type f | grep \. | rev | cut -d. -f1 | rev | sort | uniq
css
done
gif
htaccess
html
jpg
js
map
png
svg
txt
xml
```

aallan, 2021-08-17:
> @aallan How often are we likely to be changing the CSS files? (i.e. how necessary is this change?)

Realistically, once things settle down, I don't see making changes that often if at all?

mudge, 2021-08-17:
Cloudflare will cache files with the following extensions by default (note it doesn't cache `html`):

* avif
* bmp
* ejs
* jpeg
* pdf
* ps
* ttf
* class
* eot
* jpg
* pict
* svg
* webp
* css
* eps
* js
* pls
* svgz
* woff
* csv
* gif
* mid
* png
* swf
* woff2
* doc
* ico
* midi
* ppt
* tif
* xls
* docx
* jar
* otf
* pptx
* tiff
* xlsx

andrum99, 2021-08-17:
Would it not be easier to just send cache control headers for all the content on the site? That way you just have to change one setting, i.e. add a couple of extra header fields. See https://github.com/raspberrypi/documentation/pull/2026#issuecomment-899502390

andrum99, 2021-08-17:
> Realistically, once things settle down, I don't see making changes that often if at all?

Famous last words 😂

lurch, 2021-08-17:
> Realistically, once things settle down, I don't see making changes that often if at all?

Yeah, that's what I was kinda assuming. So perhaps the overly-keen caching is only a minor inconvenience, and it's not worth the effort of adding this to the toolchain? :man_shrugging:  It's probably sufficient to just know that the online-version of the site will "eventually" look the same as the local `make serve_html` version?

> Cloudflare will cache files with the following extensions by default
> * midi

Oooh, can we have background-music playing when you visit the Raspberry Pi Documentation site @aallan ? :laughing: :notes: :pray:

aallan, 2021-08-17:
> It's probably sufficient to just know that the online-version of the site will "eventually" look the same as the local make serve_html version?

On balance it feels like this would be a lot of work for not all that much gain?

Cloudflare will cache files with the following extensions by default

> > midi
>
> Oooh, can we have background-music playing when you visit the Raspberry Pi Documentation site @aallan ? 😆 🎶 🙏

No, no we may not.

mudge, 2021-08-17:
> Would it not be easier to just send cache control headers for all the content on the site? That way you just have to change one setting, i.e. add a couple of extra header fields. See [#2026 (comment)](https://github.com/raspberrypi/documentation/pull/2026#issuecomment-899502390)

We could explicitly set a shorter `max-age` but we probably _do_ want the CSS and JavaScript to be cached for as long as possible by both Cloudflare and the user's browser. In that case, the only way to signify to the cache that the file has updated is to change its URL (hence the fingerprinting you see in most web frameworks).

For this particular site, I think it's reasonable to say that styling/scripting changes are rare enough that we can allow changes to take a while to propagate and we can always explicitly purge the Cloudflare cache ourselves if need be.

aallan, 2021-08-17:
> For this particular site, I think it's reasonable to say that styling/scripting changes are rare enough that we can allow changes to take a while to propagate and we can always explicitly purge the Cloudflare cache ourselves if need be.

I think I agree. Good to get a discussion, but I don't see any big wins here for us.

closed 2021-08-17 by aallan