raspberrypi/documentation #2095: Default branch problem?

aallan, 2021-08-18:
This is **not** the same problem we saw before (pre-launch) involving PR notifications. However, when you switch Github to display the `master` rather than the `develop` (default) branch you see this notification at the top of the screen.

[image]

These 38 commits (at time of writing) are the 38 "Push to Production" merge commits,

[image]

while I can totally see why it's doing this, it seems like a UI bug rather than a thing it should be tell us?

What do you think @martinwoodward?

lurch, 2021-08-18:
We've now fixed up a slight snafu that resulted in the "Files changed: 1" in that screenshot, but https://github.com/raspberrypi/documentation/tree/master still displays "This branch is 40 commits ahead of develop.".

aallan, 2021-08-18:
> We've now fixed up a slight snafu that resulted in the "Files changed: 1" in that screenshot, but https://github.com/raspberrypi/documentation/tree/master still displays "This branch is 40 commits ahead of develop.".

Yes, was wondering if the cherry-pick would fix it. It did not.

aallan, 2021-09-07:
Still wracking up diverging "push to production" commits. 

[image]

closed 2021-10-07 by aallan