raspberrypi/documentation #2099: Copy to Clipboard button?

aallan, 2021-08-18:
So @lurch pointed me at this PR, balena-io/docs#2052, which adds a "copy to clipboard" button to all source blocks. I would sort of love to have this for our own documentation. 

https://user-images.githubusercontent.com/22801822/129884408-6a35aa25-089f-44e8-a4cb-834374c11b66.mp4

Not sure we can directly copy the original code as it's under Apache 2.0 which imposes some onerous terms with respect to redistribution on us which will conflict with our own code which is under BSD 3-Clause.

aallan, 2021-08-18:
@nelliemckesson Any thoughts?

lurch, 2021-08-18:
Oooh, I never noticed before, but multi-line code-blocks in GitHub issues
```
like
  this
      one...
```
also have a "Copy to clipboard" button that appears at the top-right when you mouse-over them! :slightly_smiling_face:

vipulgupta2048, 2021-08-23:
Hey folks, having a "Copy to clipboard" button would be a great idea for the RPi documentation. To be frank, my implementation is referenced on a snippet an ex-colleague found: https://gist.github.com/yangshun/55db997ed0f8f4e6527571fc3bee4675 here. 

I actually loved how GitHub did the "Copy to clipboard" button and wanted my implementation & experience to be similar to that. I added hover to my Clipboard button as I didn't want the actual code snippet to be hidden as @lurch observed above. I used [primer](https://primer.style/css/components/tooltips) for the tooltips and [ClipboardJS](https://clipboardjs.com/) for the actual copying. Hope this helps, y'all!

PS: Absolutely loved the work @aallan and your team did with the [new RPi documentation](https://www.raspberrypi.org/blog/bring-on-the-documentation/). As a fellow documentarian, nothing brings me more joy than seeing good people work on docs with a passion to continuously improve the experience of the users one PR at a time ✨  
Keep up the good work.

andrum99, 2021-09-06:
LGTM on Chrome and Edge on Windows 10. I get no visual indication with the mouse not in the code block. If I mouseover the black code block I get the clipboard icon appear in the top right corner of the black block, and if I mouseover that it adds `Copy to Clipboard` to the right of the icon. Clicking copies to clipboard successfully, and the text changes to `Copied!`.

[image]

Might have been a caching problem, or perhaps it's platform specific? Try CTRL+F5 to force your browser to go fetch the latest version of all files from the server(s).

andrum99, 2021-09-06:
Also LGTM on Chrome on Android phone - icon appears when code block "clicked" by tapping on touchscreen. Tapping on icon copies to clipboard successfully and `Copied!` appears to right of icon.

aallan, 2021-09-06:
Ahhhh! You're right, it was a caching problem. My other computers, where I haven't been working on the documentation site, are showing the buttons fine. 

[image]

It's just my work laptop that isn't (despite flushing my browser cache). Presuming it's just going to resolve itself now (at some point!). So closing. Thanks for double checking!

closed 2021-09-06 by aallan