raspberrypi/documentation #2101: [NOTE] versus NOTE: style differences

aallan, 2021-08-18:
If we do,

```
[NOTE]
=======
foobar
=======
```

we get this,

[image]

but if we do

```
NOTE: foobar
```

we get this,

[image]

which looks better and doesn't have the excess whitespace at the bottom of the box.

lurch, 2021-08-18:
Hmmm, https://docs.asciidoctor.org/asciidoc/latest/blocks/admonitions/ implies that they ought to be equivalent? I wonder at which stage of the processing-pipeline the discrepancy is creeping in?

lurch, 2021-08-18:
Haha, and I guess you actually meant "[IMPORTANT] versus IMPORTANT: style differences" ? :wink:

aallan, 2021-08-18:
> Haha, and I guess you actually meant "[IMPORTANT] versus IMPORTANT: style differences" ? :wink: 

Sure. But I figure the problem is generic to all the call out blocks.

lurch, 2021-08-18:
> I wonder at which stage of the processing-pipeline the discrepancy is creeping in?

Out of curiosity I just added one of each and then looked at the HTML that comes out of the other end, and the only difference seems to be that
```
NOTE: foobar
```
results in
```
...
<td class="content">
foobar
</td>
...
```
whereas
```
[NOTE]
=======
foobar
=======
```
results in
```
...
<td class="content">
<div class="paragraph">
<p>foobar</p>
</div>
</td>
...
```

So it must be the extra `<div>` and `<p>` that are adding the excess whitespace.

PR #2129 (2021-08-31): Fixing paragraph margin for different styles of admonitions
Fixes #2101
closed 2021-08-31 by aallan