raspberrypi/documentation #2104: Incorrect tables in SPI Hardware documentation

e107steved, 2021-08-20:
At https://www.raspberrypi.org/documentation/computers/raspberry-pi.html#spi-hardware there are two tables related to SPI mappings only available on the compute modules (SPI0, SPI2), which are currently incomprehensible.
It looks as if the tables need to be converted to three columns; then maintain the current text order to repopulate the cells.

andrum99, 2021-08-20:
Well spotted. That's partly my fault. These tables were all completely broken when the documentation moved to the new toolchain (see #1911). When I noticed they were broken I attempted to fix them, but failed to notice that some were meant to be 3 columns, some 4. I will do some further tweaking.

e107steved, 2021-08-20:
Incidentally, I couldn't see any reference to the SPI0 and SPI2 alternate mappings in the CM4 data sheet (specifically the 'Alternative Function Assignments' table section 2.5.1). So either the generic documentation needs a note to say the additional mappings aren't available on the CM4, or the CM4 data sheet needs an update to both the alternate function table and the pinout listing.

andrum99, 2021-08-20:
I've filed a PR, #2107, which fixes this.

andrum99, 2021-08-20:
> Incidentally, I couldn't see any reference to the SPI0 and SPI2 alternate mappings in the CM4 data sheet (specifically the 'Alternative Function Assignments' table section 2.5.1). So either the [...]

You should file a separate issue for that - best to keep each GitHub issue to a single issue, if you see what I mean.

commit 0494e76e (2021-08-20): Merge pull request #2107 from andrum99/patch-4

Fix #2104
commit 18d3aed2 (2021-08-20): Fix #2104
closed 2021-08-20 by aallan