raspberrypi/documentation #2105: Hardware/software information for UARTs

e107steved, 2021-08-20:
A section similar to https://www.raspberrypi.org/documentation/computers/raspberry-pi.html#spi-hardware is needed to cover UART hardware.
And a section similar to https://www.raspberrypi.org/documentation/computers/raspberry-pi.html#spi-software to cover UART software.

e107steved, 2021-08-20:
I've subsequently found some of the information I was looking for here: https://www.raspberrypi.org/documentation/computers/configuration.html#configuring-uarts - but not all. In particular, it doesn't mention modem control signals.
So possibly new sections as originally suggested, with more detail than already exists, and cross-references.

lurch, 2021-08-20:
IMVHO, those tables of pinouts might be better served by something like https://pinout.xyz and https://datasheets.raspberrypi.org/bcm2835/bcm2835-peripherals.pdf & https://datasheets.raspberrypi.org/bcm2711/bcm2711-peripherals.pdf ?
Although it's obviously a decision for @aallan to make...

e107steved, 2021-08-20:
Each to their own, I think - personally I often end up compiling tables like that from disparate (and sometimes conflicting!) sources of information. So I'd be pleased to see a single source.
Incidentally, assuming information is added, I'm particularly interested in confirming the capabilities of the RTS signal for controlling an RS-485 transceiver.

e107steved, 2021-08-20:
....and to provide the information for SPI and not for UARTs would surely be inconsistent?

lurch, 2021-08-20:
> from disparate (and sometimes conflicting!) sources of information.

AFAIK, in the event of conflicting information the two datasheet PDFs I linked to above ought to be taken as definitive (although be aware of the [errata](https://elinux.org/BCM2835_datasheet_errata) for the BCM2835 datasheet).

closed 2021-10-31 by github-actions[bot]