raspberrypi/documentation #2122: Inconsistency/inaccuracy in fixed focus distance?

leoshmu, 2021-08-23:
Hello, 

I wanted to double check something. In the Official Raspberry PI Camera Guide (https://magpi.raspberrypi.org/books/camera-guide/pdf), the v2 module is listed as being in focus from 50cm to infinity, implying a hyperfocal distance of 1m. However in the documentation table on this github repo, the table lists a focus of 1m to infinity for the v1 camera, and presumably that number applies to the v2 camera too. That would imply a hyperfocal distance of 2m.

Should the table in the repo list hyperfocal distance =1 m and then both sources are consistent?

Thanks for considering

aallan, 2021-08-24:
Ping @davidplowman.

aallan, 2021-08-24:
@leoshmu Is this answered by #2102 by any chance?

leoshmu, 2021-08-24:
@aallan thanks for the suggestion, but I don't think so, I didn't see an update to the documentation regarding fixed focus distance in that commit. In that commit the camera specification is here: https://github.com/davidplowman/documentation/blob/new-camera-doc/documentation/asciidoc/accessories/camera/camera_hardware.adoc and it still says fixed focus 1m to infinity for v1, a and a blank for v2 (i think indicating that v1 and v2 are same), and then something else for HQ camera. 

The issue I'm wondering about is that this documentation disagrees with the official camera module book published by raspberry pi foundation, which in chapter 14 states that the focus is 50cm to infinity. 

The stock position of the lens is I think in the hyperfocal position - for a lens focused at hyperfocal distance H, the image will be in focus at a distance H/2 to infinity. That is why I'm wondering if either the official camera module book incorrectly assumed that the hyperfocal distance was 1m and concluded that therefore focus is 50cm to infinity, or if this repository wrote 1 m to infinity fixed focus when it meant to say 1m hyperfocal distance. 

Your response is much appreciated, hoping we can figure this out!

Best, 
Leo

davidplowman, 2021-08-24:
Hi Leo, yes I agree that it's not quite consistent, so we should sort that out. As you have seen, we're doing a lot of work on documentation, though it's more a case of writing new stuff than fixing old stuff, but we'll put this on the list of things to do!

leoshmu, 2021-08-24:
Thanks David, 
No rush at all. One thing that I think is curious is that the equations for hyperfocal distance may help provide some insight. Hyperfocal distance can be approximated by H = f^2/(Fc)+f where f is focal length, F is f stop, and c is circle of confusion diameter. If the circle of confusion diameter is assumed to be 3 pixels (as was suggested on the raspberry pi boards (https://www.raspberrypi.org/forums/viewtopic.php?t=46637#p366887), then for the v2 cameras (f = 3.04mm, F = 2.0, c = 1.12µm *3 ) the hyperfocal distance is 1.38 m, so we would be in focus from 69 cm to infinity. If the circle of confusion is 2 pixels, then hyperfocal distance is 2.06m, so we would be in focus from 1.03 m to infinity. 

This calculation demonstrates a major challenge with this kind of documentation - what does it mean to be in focus. The circle of confusion is one way to define it, in essence it is saying something is in focus as long as it doesn't have more than 2 pixels of blur or 3 pixels of blur (for our 2 pixel and 3 pixel diameter circle of confusion calculations above)

What is also interesting is to do this calculation for both v1 and v2 cameras

for the v1 camera f = 3.6 mm, F = 2.9 and pixel size is 1.4µm. So for a 2 pixel sized circle of confusion, H = 1.59 m, with focus from 80cm to infinity. Meanwhile for a 3 pixel sized circle of confusion H = 1.06 m and we are in focus 53 cm to infinity. To me this suggests that the fixed focus distances of the v1 and v2 cameras are actually different - they have different f, F, and pixel size values. 

Here is a summary:

|                                      | v1    | v2     |
|--------------------------------------|-------|--------|
| f                                    | 3.6mm | 3.04mm |
| F                                    | 2.9   | 2.0    |
| px                                   | 1.4µm | 1.12µm |
| Hyperfocal distance (2px CoC based)  | 1.59m | 2.06m  |
| Hyperfocal distance (3 px CoC based) | 1.06m | 1.38m  |

I suspect you have lines of communication with people at the RaspberryPi foundation that are experts in the camera module who know a lot more than me - I'd love to talk to them if you think any of these point are worth discussing.

Best, 
Leo

davidplowman, 2021-11-01:
I've submitted a change which clarifies a little what's in the official documentation page, though as has been pointed out, what it means to be "in focus" is a bit subjective.

I'm rather suspicious of what I see in the Camera Guide book. More than once it says things like "the Pi NoIR Camera Module has a fixed focal length of 50 cm and depth of field of 50 cm to infinity" which doesn't really make sense to me. Of course, there are quite a lot of changes planned for the camera stack in future, so the whole thing will need re-visiting at some point.

aallan, 2021-11-01:
I've just merged #2261.

closed 2021-12-01 by aallan