raspberrypi/documentation #2125: CM4 data sheet - document additional pin functions

e107steved, 2021-08-25:
According to https://www.raspberrypi.org/documentation/computers/raspberry-pi.html#spi-hardware, the CM4 (along with other CMs) supports SPI0 and SPI2 on GPIOs 35-45.
However there is no mention of where these pins appear on the connectors in the CM4 data sheet (12th July build)

aallan, 2021-08-25:
Ping @dp111

lurch, 2021-08-25:
That section also says "SPI2, also with three hardware chip selects, is only available on Compute Module 1, 3 and 3+." ?
@andrum99 did some work on this recently in #2001

e107steved, 2021-08-25:
Must be in an unpublished version - I checked immediately before posting this issue. Still leaves SPI0, though

aallan, 2021-08-25:
> Must be in an unpublished version

That should be live?

dp111, 2021-08-25:
Andrew is correct  "SPI2, also with three hardware chip selects, is only
available on Compute Module 1, 3 and 3+." maybe your cache needs refreshing?

There is a table heading which could be misleading :
"SPI2 (compute modules only)"

Maybe it should be "SPI2 (compute modules only, except CM4)"

aallan, 2021-08-25:
Updated the table header on `develop` will go to the live site next time I push to `master`.

commit 83fe238f (2021-08-25): Update spi-bus-on-raspberry-pi.adoc

Fixed #2125
closed 2021-08-25 by aallan