raspberrypi/documentation #2144: building kernel documentation lacks updating information

twoseascharlie, 2021-09-03:
I had a custom built kernel and recently did a sudo update/sudo full-upgrade. My raspberry pi would not boot afterward. Looking at the kernel output, it appeared as though it had blown out my custom kernel/modules. I did not look into it very hard and just ended up doing a fresh install.

I tried opening a forum post [(https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=318917)](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=318917) to ask how you are supposed to keep your system up to date if you've built a custom kernel and one person suggested I update the docs. I don't have enough knowledge to do so, so I am opening an issue here about it.

I think the raspberry pi documentation should have information on how to keep your system up to date if you have built a custom kernel. A concrete list of steps would be nice.

ex:
```
# update running system
sudo apt update
sudo apt full-upgrade

# update latest kernel source and build
cd <linuxKernelRepo>
git pull
# generate default config and re-apply whatever you had added to it
# build kernel
# restart
```

Any other steps or things that could help others who are new to this would be great to have in the docs. Or in the very least a warning that the usual update process should not be used, or may cause errors going forward. There is currently only a small mention about renaming your custom built kernel.

ShaileshKumar007, 2021-09-04:
Hi. I want to resolve this issue by updating the raspberry pi docs by adding the documentation on the topic "Building your own kernal"

twoseascharlie, 2021-09-06:
@ShaileshKumar007 my opinion is that this should be worked on by people who have experience building custom kernels for the raspberry pi and know what kind of implications it has on keeping the pi up to date. There may be more than one set of steps required depending on the various use cases that could account for building your own kernel. If that is you, then that would be great, but if not, then it may be best to wait for someone who has more experience to chime in.

ShaileshKumar007, 2021-09-06:
I think you are right. It will be better if an experienced person will work on this issue.

lurch, 2021-09-08:
Probably also needs some `apt` config tweaks, so that an `apt upgrade` doesn't overwrite your custom-built kernel with the distro-supplied kernel? But that's probably @XECDesign 's area of knowledge.

XECDesign, 2021-09-09:
You could `apt-mark hold package_name` to prevent it from upgrading.

JamesH65, 2021-11-14:
Ping. Think this should be kept open - its a valid request/need.

twoseascharlie, 2021-11-15:
> Ping. Think this should be kept open - its a valid request/need.

Thanks James, hopefully someone can update the kernel docs with more info.

JamesH65, 2022-01-16:
Ping, this should not be forgotten.

closed 2022-04-03 by github-actions[bot]