raspberrypi/documentation #2178: BCM2835-ARM-Peripherals.pdf and other datasheets missing

cleverca22, 2021-10-01:
https://github.com/raspberrypi/documentation/blob/develop/documentation/redirects/datasheets.csv has setup redirects to them
but the files themselves are no longer in this repo

https://datasheets.raspberrypi.org/bcm2835/ directory indexing is also disabled on the new location
so there is no way to discover what other datasheets exist, and no way to view any history of how they have changed

i would prefer if they are available in some git repo, so history is maintained, and its easy to list every datasheet

lurch, 2021-10-01:
https://datasheets.raspberrypi.org/ seems to be what you're looking for? :wink:  Some of the datasheets include a release history, but not all.
Note also that the conformity-section of the documentation now lives in https://pip.raspberrypi.org/

aallan, 2021-10-01:
The files currently (temporarily) live on http://datasheets.raspberrypi.org. However they will eventually be moving to the new Product Information Portal (PIP), http://pip.raspberrypi.org. Going forward they won’t be kept here in Git.

closed 2021-10-01 by aallan