raspberrypi/documentation #2183: Document `gpio=...` stanzas could be used in EEPROM bootloader configuration

RomanValov, 2021-10-04:
I found on raspberry pi forums that following config is valid for RPi4 EEPROM and works:

```
[all]
BOOT_UART=1
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
ENABLE_SELF_UPDATE=1

gpio=21=pu

[gpio21=1]
BOOT_ORDER=0xf2

[gpio21=0]
BOOT_ORDER=0xf1
```

However I wasn't able to find any statements in docs that `gpio=21=pu` should work in this context.

I.e. for gpio filters there is a statement [here](https://www.raspberrypi.org/documentation/computers/raspberry-pi.html#configuration-properties):

```
Conditional filters are also supported except for EDID.
```

Also there is a statement about `[config.txt]` section [here](https://www.raspberrypi.org/documentation/computers/raspberry-pi.html#config_txt).

Whereas `[config.txt]` section works for arbitrary parameters.
It's not possible to use arbitrary config.txt parameters in `[all]` section, i.e. `cmdline`.
But `gpio=...` works.

lurch, 2021-10-05:
ping @timg236 & @peterharperuk

peterharperuk, 2021-10-05:
I might be wrong but they're two different things...

gpio=21=pu

This is a command not a filter. Set pin 21 to have a pull up

[gpio21=1]
BOOT_ORDER=0xf2

This is a filter. If gpio21 is high then set the boot order to 0xf2

timg236, 2021-10-05:
Gpio pull-ups etc are supported.  But EDID and include statements are not

closed 2021-12-19 by github-actions[bot]