raspberrypi/documentation #2184: Raspberry Pi Pico Python SDK: default SPI pins incorrect

scruss, 2021-10-05:
Table 3.7, page 16 states the default pins for SPI(0) are as follows:

    SPI0_SCK      Pin 6
    SPI0_MOSI     Pin 7
    SPI0_MISO     Pin 4

But in the REPL, one sees:

    MicroPython v1.17 on 2021-09-02; Raspberry Pi Pico with RP2040
    Type "help()" for more information.
    >>> from machine import SPI
    >>> s=SPI(0)
    >>> s
    SPI(0, baudrate=992063, polarity=0, phase=0, bits=8, sck=18, mosi=19, miso=16)

So table 3.7 should read:

    SPI0_SCK      Pin 18
    SPI0_MOSI     Pin 19
    SPI0_MISO     Pin 16

The entries for SPI(1) appear to be correct.

aallan, 2021-10-05:
> Table 3.7, page 16 states the default pins for SPI(0) are as follows:

...of which book?

There is a difference between the MicroPython default pins and the Pico C SDK default pins, which are the "default" pins used everywhere in the documentation.

scruss, 2021-10-05:
The book called "Raspberry Pi Pico Python SDK" at https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-python-sdk.pdf

scruss, 2021-10-05:
and yet after a Pico reboot (which I'd done before):

    MicroPython v1.17 on 2021-09-02; Raspberry Pi Pico with RP2040
    Type "help()" for more information.
    >>> from machine import SPI
    >>> s=SPI(0)
    >>> s
    SPI(0, baudrate=992063, polarity=0, phase=0, bits=8, sck=6, mosi=7, miso=4)

back to the *default* defaults. No idea how this could have come about. Closing, not reliably repeatable.

closed 2021-10-05 by scruss

lurch, 2021-10-05:
How odd! But FYI https://github.com/micropython/micropython/blob/master/ports/rp2/machine_spi.c#L43 agrees with the table in https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-python-sdk.pdf