raspberrypi/documentation #2254: Pi Zero 2 W: Conditional model filters in config.txt

dwhinham, 2021-10-30:
Hi there!

Is there a conditional `config.txt` filter that will specifically capture the new Pi Zero 2 W?

Using `[pi0]` or `[pi02w]` seems to be working for me but I'm not sure if the latter is correct.
Either way it'd be great if this could be documented in the table shown at: https://www.raspberrypi.com/documentation/computers/config_txt.html#model-filters

Many thanks!

aallan, 2021-10-31:
Ping @ghollingworth

ghollingworth, 2021-10-31:
@pelwell

pelwell, 2021-10-31:
`[pi02]` is the model-specific section. The Zero 2 is also included by `[pi0]` sections.

aallan, 2021-10-31:
Added in 12c3c2a, pushed to production in #2257.

closed 2021-10-31 by aallan

dwhinham, 2021-10-31:
Thanks for the fast responses everyone, much appreciated! :tada:

Really enjoying the new model. :slightly_smiling_face: