raspberrypi/documentation #2256: AsciiDoc and Markdown handle backticks differently

pelwell, 2021-10-31:
As the title says, AsciiDoc doesn't handle backticks in quite the same way as Markdown. This is noticeable in the Device Tree documentation, where some node names have underscores at both ends - `__overlay__`, `__dormant__` and `__fixups__`, for example. Markdown sees the characters between the backticks as literal strings to be displayed monospaced on a grey background and with no further interpretation, whereas AsciiDoc also interprets the underscores as requesting emphasis, so the underscores disappear and the remaining text is italicised (I'm not sure how I would code that in Markdown).

Here is the AsciiDoc source for the worst section: https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/configuration/device-tree.adoc#overlayfragment-parameters

And here is how it appears on the website: https://www.raspberrypi.com/documentation/computers/configuration.html#part2.2.8

Note that the AsciiDoc rendering by GitHub and by the website are in agreement.

pelwell, 2021-10-31:
Ah - `+` quoting can be used to avoid the underscores being interpreted: 
```
`+__overlay__+`
```

pelwell, 2021-10-31:
And while you're in there, you can update the first paragraph to say:

> The DT parameter mechanism as described has a number of limitations, including no easy way to create arrays of integers and the inability to create new nodes. One way to overcome some of these limitations is to conditionally include or exclude certain fragments.

Thanks.

aallan, 2021-10-31:
Thank @pelwell. I'll try and get that sorted tomorrow.

lurch, 2021-10-31:
```bash
$ git grep -l '__' "*.adoc"
documentation/asciidoc/accessories/camera/timelapse.adoc
documentation/asciidoc/computers/configuration/device-tree.adoc
```

commit b11e0aca (2021-11-04): Fix display of double-underscores inside backticks.
Fixes #2256
PR #2267 (2021-11-04): Fix display of double-underscores inside backticks.
Fixes #2256

ping @pelwell
closed 2021-11-04 by aallan