raspberrypi/documentation #2271: Display can now work as separate monitor

duckida, 2021-11-05:
https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/accessories/display/display_intro.adoc says that the touch display cannot work as a separate X monitor, but with Pi 4 and Buster, it now can. I have tested, and it has worked.

lurch, 2021-11-05:
ping @6by9 for clarification.

aallan, 2021-12-01:
> says that the touch display cannot work as a separate X monitor

I've just re-read this page, it seems to say the opposite. The opening paragraphs says, _"The Raspberry Pi Touch Display is an LCD display which connects to the Raspberry Pi through the DSI connector. In some situations, it allows for the use of both the HDMI and LCD displays at the same time…"_ so I'm not sure what you mean here? Can you clarify?

JamesH65, 2021-12-01:
You cannot have to independent displays (ie two version of X running, although not sure that is useful in any way), but you can mix DSI and HDMI on Buster (FKMS) and Bullseye (KMS). On KMS you can actually have 2xHDMI  AND the DSI display all rendering various parts of the desktop.

aallan, 2021-12-01:
> You cannot have to independent displays (ie two version of X running, although not sure that is useful in any way)

Two independent X Server instances? Nope, never thought that was possible.

> but you can mix DSI and HDMI on Buster (FKMS) and Bullseye (KMS). On KMS you can actually have 2xHDMI AND the DSI display all rendering various parts of the desktop.

That's what I thought was going as well. Does your reading of the current documentation parse like that, because it does for me?

JamesH65, 2021-12-01:
Mostly OK, FKMS should say "KMS and FKMS", and the diagram at the bottom of the page should be removed as its not valid for all models of display.

aallan, 2021-12-01:
Just removed the diagram (on develop). All instances of FKMS should be replaced?

JamesH65, 2021-12-01:
Hmm, I think  should say "FKMS and KMS", as we still support FKMS, KMS only became the default on Bullseye, so very recent, and it can be easily turned off to get round Bullseye bugs.

aallan, 2021-12-01:
How's that @JamesH65 ?