raspberrypi/documentation #2285: Docs website inconsistent scrolling area for index

JamesH65, 2021-11-12:
[image]

When mouse is over the index panel, the mouse wheel scrolls the index if you over text (the red araa in picture) butt he main window in other places, so a very inconsistent user experience, a slight change in mouse position and suddenly a completely different windows starts scrolling.

aallan, 2021-11-12:
> When mouse is over the index panel, the mouse wheel scrolls the index if you are over text (the red area in picture) but the main window in other places

When the mouse is over the index it scrolls the index.

When the mouse is over the main panel it scrolls the main panel. 

Not sure what the problem is?

lurch, 2021-11-12:
I think James is expecting the index to still scroll if the mouse-cursor is to the *left* of the index-area.

JamesH65, 2021-11-12:
That is indeed what I expect.

aallan, 2021-11-12:
That's counterintuitive to me, if you're not over the index the main panel should be the one that scrolls.

JamesH65, 2021-11-12:
You are over the index, just not over the textual part of the index. Mouse scroll over the text, then move the mouse *left* off the text but still in the index. Stop scrolling, start again, suddenly the other window starts scrolling.

Perhaps I should add this is on Chrome on Windows. Might behave differently on other devices.

JamesH65, 2022-01-16:
Ping. this should not be forgotten.

JamesH65, 2022-04-03:
Reopening as this is still very odd behaviour which should be fixed.

aallan, 2022-04-24:
This isn't going to get fixed.

closed 2022-04-24 by aallan

JamesH65, 2022-04-24:
Shame, it's really odd behaviour that should be fixed.