raspberrypi/documentation #2297: Documentation PDF Files Use Placeholder Colophon

Marsfan, 2021-11-21:
A lot of the Documentation PDF files do not have a colophon set in them, instead they just use the placeholder colophon. 

For an example, see the start of page 2 of the CM4 datasheet.
https://datasheets.raspberrypi.com/cm4/cm4-datasheet.pdf

aallan, 2021-11-21:
I'm not really sure what you mean by "placeholder colophon" ..?

Marsfan, 2021-11-22:
Here is the an example from the CM4 datasheet. The bit word **Colophon** should probably not be there.

<details> 
<summary> Missing Colophon </summary>

[image]

</details>

As a side note, I had sent an email in through the contact form on the Raspberry Pi website and was told to put the issue here instead.

aallan, 2021-11-22:
You're arguing that the Colophon shouldn't have a section header "Colophon" in the books?

Marsfan, 2021-11-24:
I may be mistaken, but I believe that instead of saying "Colophon", it should have some sort of publisher's wwordmark. 
Feel free to close if I am mistaken.

closed 2021-11-24 by aallan