raspberrypi/documentation #2305: obsolete command line option "elevator"

kaese, 2021-11-25:
command line option "elevator" is obsolet:

[    0.000000] Kernel parameter elevator= does not have any effect anymore.
               Please use sysfs to set IO scheduler for individual devices.

See also https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=971882

It should be removed from documentation section "Command Line Options".

lurch, 2021-11-25:
ping @pelwell @XECDesign

aallan, 2021-11-25:
This option seems to be mentioned three times in the current documentation,

```
$ grep -R "elevator" *
asciidoc/computers/configuration/kernel-command-line-config.adoc:* elevator: specifies the I/O scheduler to use. `elevator=deadline` means the kernel imposes a deadline on all I/O operations to prevent request starvation.
asciidoc/computers/remote-access/network-boot-raspberry-pi.adoc:root=/dev/nfs nfsroot=10.42.0.211:/nfs/client1,vers=4.1,proto=tcp rw ip=dhcp rootwait elevator=deadline
html/computers/remote-access.html:<pre>root=/dev/nfs nfsroot=10.42.0.211:/nfs/client1,vers=4.1,proto=tcp rw ip=dhcp rootwait elevator=deadline</pre>
html/computers/configuration.html:<p>elevator: specifies the I/O scheduler to use. <code>elevator=deadline</code> means the kernel imposes a deadline on all I/O operations to prevent request starvation.</p>
$
```

lurch, 2021-11-25:
Only twice - the output files in `html/*` get autogenerated from the source files in `asciidoc/*` :wink: 
(personally, I like to avoid this issue by doing `git grep ...` instead of a regular `grep -R ...`)

aallan, 2021-11-25:
> Only twice - the output files in html/* get autogenerated from the source files in asciidoc/* 😉

Doh! Yes, sorry! 🤦‍♂️

pelwell, 2021-11-25:
Given that the command line option was obsoleted in 5.0 (sic) it should be deleted from the documentation.

aallan, 2021-11-25:
Thanks @pelwell! 

In the case of the network boot instance, just drop it from the incantation? Everything else remains the same?

pelwell, 2021-11-25:
Yup.

aallan, 2021-11-25:
Thanks all. Fixed on `develop` will go live next time I push the `master` branch to production.

PR #2306 (2021-11-25): Remove obsolete kernel option
Fixes #2305
closed 2021-11-25 by aallan