raspberrypi/documentation #2310: KMS, FKMS, and Legacy

aallan, 2021-12-01:
Need an explanation of the three different graphics stacks, see [the discussion](https://github.com/raspberrypi/documentation/pull/2309#discussion_r760280457) in #2309.

aallan, 2021-12-01:
@JamesH65 says:
> FKMS is "Fake" KMS, it provides the KMS API to 2D graphics (HW composition etc, NOT 3D) and display, but instead of controlling the hardware, it makes calls in to the firmware on the Videocore which in turns does the HW access. In KMS, all the HW accesses are made from ARM space, so the firmware is not involved. It means the entire thing is now OSS, and all control of attached HW like DPI and DSI displays is in the kernel space, so people can now write their own drivers etc.
>
> Legacy is where everything is controlled by the firmware, no standard KMS API. The firmware takes complete control of all the hardware (HVS, 3D etc) and this is accessed using fairly non-standard mechanisms, via VCHIQ and some ARM side shim libraries.
Sort of feels like we should have a section talking about this somewhere between "[HDMI Configuration](https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/configuration/hdmi-config.adoc)" and "[Rotating your Display](https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/configuration/display-rotation.adoc)" in [configuration.adoc](https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/configuration.adoc)?
@lurch says:
> With maybe a nice picture explaining how things fit together? And could maybe also explain how different Pis and different OS versions default to using different graphics stacks? 🤷

aallan, 2022-02-07:
Still needs doing.

lurch, 2022-03-03:
Latest word from @spl237 is that FKMS is no longer supported (?) so I guess that cuts this down to just KMS and Legacy.

cletusw, 2022-04-29:
> Latest word from @spl237 is that FKMS is no longer supported (?) so I guess that cuts this down to just KMS and Legacy.

Ubuntu is [still (26 Apr 2022) recommending FKMS](https://ubuntu.com/blog/whats-new-for-raspberry-pi-ubuntu-22-04-lts) as the supported way to use the official [Raspberry PI Touch Display](https://www.raspberrypi.com/documentation/accessories/display.html) with Ubuntu. Is there a migration path or is this setup no longer supported?

JamesH65, 2022-04-29:
@6by9 AIUI, the Pi LCD is now fully supported by KMS?

It will still work under FKMS.

lurch, 2022-05-03:
The documentation (in this repo) is only about Raspberry Pi OS - it's up to Ubuntu to document their own OS :wink: 

Questions about "is this setup no longer supported" probably belong in https://github.com/raspberrypi/firmware ?

cletusw, 2022-05-03:
Good point! Sorry for the confusion.

6by9, 2022-05-03:
> AIUI, the Pi LCD is now fully supported by KMS?

Yes, the DSI display is fully supported under KMS.

> It will still work under FKMS.

Should do, but FKMS now only receives limited testing.

aallan, 2023-04-27:
This is also going to need updating ahead of bookworm. Should probably be done on [the bookworm branch](https://github.com/raspberrypi/documentation/tree/bookworm) at this point.

aallan, 2023-05-15:
From Simon via @lurch 

> In terms of graphics stacks, FKMS is now totally gone, Wayland only targets KMS, and X can target either X or KMS

From Gordon via @lurch 

> Gordon thinks that WayFire (with the KMS backend) will be the *only* supported graphics stack on bookworm, and neither X nor the Legacy graphics stuff will be supported.

6by9, 2023-05-15:
Using KMS directly or via the emulated fbdev interface also exists for non-GUI use cases.

The original discussion was only on legacy/FKMS/KMS. X and/or Wayfire sit on top of those. Which layer are you wanting documenting?

aallan, 2023-07-30:
Still a thing.

lurch, 2023-08-30:
Fixed on the `bookworm` branch.

commit dc795610 (2023-08-30): Updates to the HDMI configuration documentation
Fixes #2310 #2957 #3067
closed 2023-08-30 by lurch