raspberrypi/documentation #2314: Update documentation for differences between Bullseye and Buster

lurch, 2021-12-07:
We now [offer both](https://www.raspberrypi.com/news/new-old-functionality-with-raspberry-pi-os-legacy/) Bullseye and Buster (Legacy) releases of Raspberry Pi OS. However some features work in one but not the other, and vice-versa.

@JamesH65 has an internal document which is probably a useful starting point for this.

lurch, 2021-12-07:
And perhaps https://www.raspberrypi.com/documentation/computers/getting-started.html#installing-the-operating-system and/or https://www.raspberrypi.com/documentation/computers/os.html#introduction ought to mention the different versions too?

URRG, 2021-12-17:
I want to work on this issue so from where should i start

JamesH65, 2021-12-17:
This is already underway internally, but it's a private repo.

JamesH65, 2021-12-17:
Sorry, closed by accident.

lurch, 2022-02-21:
This should stay open.

aallan, 2022-04-24:
Should stay open. Still some updating needed. See also #2398.

RuffaloLavoisier, 2022-04-24:
@aallan 
Are you simply updating the vlc and omx examples according to the version?

aallan, 2022-04-24:
> @aallan Are you simply updating the vlc and omx examples according to the version?

OMX Player no longer works under Bullseye. All the examples using it in the documentation need to be replaced with an equivalent invocation for VLC (or other relevant software), see #2398 which is the linked issue.

However this issue is the more general "other things that Bullseye broke" issue. There are some other things beyond the OMX Player issue that haven't yet been fixed.

RuffaloLavoisier, 2022-04-24:
If there is something to be done externally, I would be happy to contribute.

lurch, 2022-07-02:
This still needs doing, at some point.

JamesH65, 2022-09-04:
ping

aallan, 2023-04-27:
Headline things:

 
* X goes away and is replaced by Wayland
   - nominally a big thing, but there has been a lot of work to keep the user experience the same
   - but some change is inevitable
* dhcpcd goes away and is replaced by Network Manager
* random bookworm stuff

aallan, 2023-04-27:
**Note:**  Documentation should be updated to reflect the *latest* version of the OS. So we're going to go through and make sure it all works with Bookworm. We may need some NOTE blocks that mention "for previous versions of Raspberry Pi OS" from time to time, but these need to be kept short (a couple of medium-length paragraphs at most). We do not have the availability to support multiple versions of the OS in the documentation going forward with breaking changes like Wayland and Network Manager happening.

aallan, 2023-04-27:
Current tasks associated with this issue:

```[tasklist]
### Bookworm Issues
- [ ] https://github.com/raspberrypi/documentation/issues/2687
- [ ] https://github.com/raspberrypi/documentation/issues/2398
- [ ] https://github.com/raspberrypi/documentation/issues/2310
```

this task list will expand rapidly when @lurch does the documentation audio.

lurch, 2023-07-04:
Child-issues:
* #2999 
* #2947 
* #2948 
* #2949 
* #2950 
* #2951 
* #2952 
* #2953

aallan, 2023-09-22:
Done.

closed 2023-09-22 by aallan