raspberrypi/documentation #2325: SD Card for Raspberry Pi

juanca1944, 2021-12-21:
"Because of a hardware limitation in the Raspberry Pi Zero, 1 and 2, the boot partition on the SD card must be 256GB or less..."

I think it must say 256MB not 256GB

Lanwalker, 2021-12-21:
I doubt that 256 ' MB '. It sure isn't on the Pi 4

Robert Gamble - Disabled IT Analyst/Communication Systems Engineer & Automation Test, Control and Design Systems Engineer, 82nd Airborne Artillery ( 1977 ), National Admin for the United States Minutemen - American Patriots Coalition, Service Officer for the Newton American Legion, East Fultonham, Ohio

***@***.*** ( preferred )
***@***.***
***@***.***
***@***.***

lurch, 2021-12-22:
Duplicate of #2263 ?

juanca1944, 2021-12-22:
Yes Lurch you are right, it is a duplicate. Sorry.
We are talking about a microSD card not a HDD. As an example I use a 2GB microSD with  DietPi and a boot partition of 128MB. 
If you use or used a Raspberry PI you must noticed that all the boot partitions are 256MB or less, even Windows, of course there must be an exception.

closed 2021-12-22 by lurch