raspberrypi/documentation #2332: Investigate incorrect anchors

lurch, 2022-01-04:
I'm not sure how #2328 "slipped through the net", so investigate it further and possibly add a script to check that there are no other incorrect anchors.

lurch, 2022-01-07:
Another occurrence of this in #2335 so we definitely need a way of automatically finding these. (and once it's working, it probably makes sense to add it to the CI checks too.)

RuffaloLavoisier, 2022-01-08:
@lurch

lurch, 2022-01-08:
Well you can either submit PRs, or you can just wait for me to write this "anchor checking script" next week, which _should_ find them all automatically. :slightly_smiling_face: 
But feel free to make a list of them in this issue, which I can use to make sure that the "anchor checking script" is working correctly.

RuffaloLavoisier, 2022-01-08:
>you can just wait for me to write this "anchor checking script" next week, which should find them all automatically. 🙂

That's great. 

Is there anything I can do to contribute?

aallan, 2022-01-08:
> Is there anything I can do to contribute? any help?

If you can list any other bad links you find in this thread, we'll clean them up when @lurch is back from holiday.

RuffaloLavoisier, 2022-01-08:
@aallan   Sure - I'll try to find it.

RuffaloLavoisier, 2022-01-09:
# Hi there
I found all the broken links in script. I'm a human being, so I may have missed something.
I'll leave a Pull Request.

```
accessories/camera/libcamera_detect.adoc
```
target keyword [line 17](https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/accessories/camera/libcamera_detect.adoc)
[`object-detect-tf-stage`](https://www.raspberrypi.com/documentation/accessories/camera.html#libcamera-detect)

```
computers/raspberry-pi/bcm2711-bootloader.adoc
```
target keyword [line 417](https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/raspberry-pi/bcm2711-bootloader.adoc)
[`fail-safe-os-updates-tryboot`](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#boot_ramdisk)

```
accessories/camera/camera_usage.adoc
```
target keyword [line 35](https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/accessories/camera/camera_usage.adoc)
[`installing-a-raspberrypi-pi-camera`](https://www.raspberrypi.com/documentation/accessories/camera.html#libcamera-and-the-legacy-raspicam-camera-stack)

## Please check me
I have a question. How is this link connected? The ID values are different, but they are the same on the site.
I couldn't find the id form anywhere.

in link : https://www.raspberrypi.com/documentation/computers/processors.html

Go to link : https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#raspberry-pi-3-model-b-2

*INPUT*

[line 3](https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/processors/rp3a0.adoc)
```
documentation/asciidoc/computers/processors/rp3a0.adoc
```
`xref:raspberry-pi.adoc#raspberry-pi-3-model-b-2`

*OUTPUT*

[line 18](https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/raspberry-pi/raspberry-pi-schematics.adoc)
```
documentation/asciidoc/computers/raspberry-pi/raspberry-pi-schematics.adoc
```

```
=== Raspberry Pi 3 Model B

* https://datasheets.raspberrypi.com/rpi3/raspberry-pi-3-b-reduced-schematics.pdf[Schematics, Revision 1.2]
* https://datasheets.raspberrypi.com/rpi3/raspberry-pi-3-b-mechanical-drawing.pdf[Mechanical Drawings, PDF]
* https://datasheets.raspberrypi.com/rpi3/raspberry-pi-3-b-mechanical-drawing.dxf[Mechanical Drawings, DXF]
```

`raspberry-pi-3-model-b-2`, where is "2" ???

lurch, 2022-01-09:
> `raspberry-pi-3-model-b-2`, where is "2" ???

That's because it's the second time that the `raspberry-pi-3-model-b` ID appears on the page, so various parts of the build-process (I can't remember where off the top of my head) add a `-2` suffix to ensure that all IDs on a page remain unique.

RuffaloLavoisier, 2022-01-09:
@lurch 
I see, Thank you

commit 833e7b2c (2022-01-12): Merge pull request #2345 from raspberrypi/check_links_and_anchors

Check that all internal links and anchors are valid. Fixes #2332
commit d0d19a76 (2022-01-12): Check that all internal links and anchors are valid. Fixes #2332
closed 2022-01-12 by aallan