raspberrypi/documentation #2337: access-point-routed: iptables

DeeJay, 2022-01-09:
Ref: [https://www.raspberrypi.com/documentation/computers/configuration.html#enable-routing-and-ip-masquerading](https://www.raspberrypi.com/documentation/computers/configuration.html#enable-routing-and-ip-masquerading)

Requires **iptables**, but iptables is not pre-installed in any of the 3 variants of the current (31Oct2021) RasPiOS Bullseye [32] image.

lurch, 2022-01-12:
ping @XECDesign  Did iptables used to be included by default in Buster but no longer included by default in Bullseye?

XECDesign, 2022-01-12:
iptables was previously included, but not explicitly. It was dragged in by something else.

Documentation should probably be updated to use nftables instead.

lurch, 2022-01-12:
> Documentation should probably be updated to use nftables instead.

Is nftables also supported in Buster, or would the instructions need to use iptables on Buster and nftables on Bullseye?

DeeJay, 2022-01-12:
> > Documentation should probably be updated to use nftables instead.
> 
> Is nftables also supported in Buster, or would the instructions need to use iptables on Buster and nftables on Bullseye?

There is no overlap. Buster comes with iptables but not nftables. Conversely, Bullseye comes with nftables but not iptables. (Determined by inspection of the .info files for each image/distro.)

XECDesign, 2022-01-13:
> Is nftables also supported in Buster,

I assume either is supported in both cases (I haven't had cause to use nftables myself yet), you'd just need to add an appropriate `apt install` step.

> or would the instructions need to use iptables on Buster and nftables on Bullseye?

Is the documentation policy that instructions need to work on oldstable as well? Seems like that could be quite painful in some situations.

aallan, 2022-01-13:
> Is the documentation policy that instructions need to work on oldstable as well? Seems like that could be quite painful in some situations.

Since the "Legacy" release is officially supported, and is currently Buster (not Bullseye) both have to be supported by the documentation.

closed 2022-03-27 by github-actions[bot]