raspberrypi/documentation #2339: Device ID incorrect for Cyan Medium Angular Motor in BuildHAT documentation

mutesplash, 2022-01-09:
I just purchased lego part 45603 with Article ID 6254347 on the invoice.  This part is known as "Medium Angular Motor" on Lego Education's site and is Cyan in color.  Using it with BuildHAT returns a Device ID of int 48 or 0x30.  This conflicts with documentation/asciidoc/accessories/build-hat/compat.adoc which lists "Large Angular Motor" as possessing this Device ID.

Unfortunately I do not have a Cyan "Large Angular Motor" to test and see if the Device IDs for these motors has been swapped, which I _suspect_ is the issue here.  Perhaps somebody who does can run this python script to confirm?

```
import buildhat
from buildhat import Hat 

hat = Hat()
print(hat.get())
```

I suppose it's possible that two devices have the same ID or that the Cyan "Medium Angular Motor" that came with the 2020 Spike PRIME set is a different motor than part 45603.

mutesplash, 2022-01-09:
I should also note that the PDF datasheet documenting the serial protocol notes (linked in documentation/asciidoc/accessories/build-hat/links-to-other.adoc) that the "SPIKE Prime medium motor" is "active ID code" is 0x30 and the "large motor" is 0x31, lending credence to my idea that the problem is that these IDs are swapped .

lurch, 2022-01-10:
ping @topshed

topshed, 2022-01-12:
@mutesplash is correct - these two IDs are the wrong way round in the compatibility list.

closed 2022-01-12 by aallan