raspberrypi/documentation #2352: NVMe documentation is out of date

peterharperuk, 2022-01-14:
documentation/computers/raspberry-pi.html#nvme-ssd-boot

It says you need pre-release software but the bullseye release and latest stable bootloader has everything you need.
I had a request to publish a list of drives that we know work. This might be a big dodgy but I'll push an update for someone to review.

lurch, 2022-01-14:
In previous years we've had similar requests for lists of known-working SD cards and known-working USB drives, but I believe we've always refrained from doing so, because:
* any such list will always be hopelessly incomplete
* what works for one person may not work for another person, e.g. if they're using a marginal power supply
* what used to "not work" might work in future, with updated bootloader / firmware / kernel
* with cheap SD cards there's also the problem of counterfeits

:shrug:

lurch, 2022-01-14:
> It says you need pre-release software but the bullseye release and latest stable bootloader has everything you need.

Does it also work out-of-the-box with (the stable bootloader and) Buster Legacy?

peterharperuk, 2022-01-14:
> Does it also work out-of-the-box with (the stable bootloader and) Buster Legacy?

Yes, and new devices should have the required bootloader installed already.

aallan, 2022-01-14:
> I had a request to publish a list of drives that we know work.

While we need to update the documentation to say "this works now" I agree with @lurch, I don't think we want to publish a list. It'll be continually out of date, and always incomplete.

aallan, 2022-01-14:
Thanks @peterharperuk !

closed 2022-01-14 by aallan