raspberrypi/documentation #2360: config.txt: hdmi_* while using dtoverlay=vc4-kms-v3d

jpc0, 2022-01-17:
I seems like the hdmi_force_hotplug, hdmi_force_mode, hdmi_group ,hdmi_mode and hdmi_drive options have no effect in linux when dtoverlay=vc4-kms-v3d is enabled, at least on the RPI 3B+ on bullseye. The only place I found this mentioned was [here](https://forums.raspberrypi.com/viewtopic.php?t=326126#p1951931) and it's pretty buried.

We should maybe include some information somewhere that indicates this and if there is an alternative way to forcing these options document that.

aallan, 2022-01-17:
Sounds like this is a sub-issue of #2310?

lurch, 2022-01-18:
Also partially overlaps with some of the internal documentation mentioned in #2314

closed 2022-03-27 by github-actions[bot]