raspberrypi/documentation #2363: looking for cmake best practices to support multiple version of RPi's

ebo, 2022-01-18:
I hope that this question is not out of line here -- I Cannot figure out where else to ask.

I am working to clean up the CMakeLists.txt in https://github.com/ebo/machinekit-hal.  Since we need to provide support for at least RPi 3&4's, as well as non-Pi platforms in the future, I am looking for best practices to find the  architecture,  model, and if the OS is 64 or 32 bits.  I found a couple of places where people use thins similar to this:

#=====================================
if("$ENV{RASPBERRY_VERSION}" STREQUAL "")
    set(RASPBERRY_VERSION 1)
else()
    if($ENV{RASPBERRY_VERSION} VERSION_GREATER 3)
        set(RASPBERRY_VERSION 3)
    else()
        set(RASPBERRY_VERSION $ENV{RASPBERRY_VERSION})
    endif()
endif()

# 3B+: ARM Cortex-A53 64bit (Armv8-A) 1.4GHz (4 cores)
# 4B: ARM Cortex-A72 (Armv8-A) 1.5GHz (4 cores)
#
# -march=armv8-a
# -mcpu=cortex-a72
# -mcpu=cortex-a53
#=====================================

but I cannot figure out what sets RASPBERRY_VERSION in the OS, and if I can find the rest of the information to set the compiler flags logically.

If I can figure out a robust way to handle this I will try and help write a documentation section with the example.

Thanks in advance.

lurch, 2022-01-18:
This sounds more like a question for the [forums](https://forums.raspberrypi.com/) ?

ebo, 2022-01-18:
@lurch, I've dug around a little in the forums, and will ask over there as well.  I'll leave this open for a little while to see if I get any other nibbles before closing it.  Thank you for the pointer.

lurch, 2022-01-18:
FYI the canonical way of figuring out "what model of Raspberry Pi am I running on?" is https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#raspberry-pi-revision-codes

andrum993, 2022-01-18:
> EDIT: And you might find the NOTE here useful https://www.raspberrypi.com/documentation/computers/configuration.html#kernel-files

That note is slightly misleading. On Raspberry Pi, `lscpu` returns information about the currently running operating system, not the hardware. For example, if a 64-bit SoC is running a 32-bit version of Linux, then lscpu will return `armv6`, `armv7` or `armv7l`. `lscpu` will only return `armv8` if the running Linux OS is also 64-bit.

closed 2022-03-27 by github-actions[bot]